arduino/ArduinoCore-megaavr #1: Uno2018 Variant pin picture is badly formatted, and doesn't match the tables.

WestfW, 2018-10-31:
There's an ascii picture of the ATmega4809 in variants/uno2018/pins_arduino.h that is horribly formatted with a mixture of tabs and spaces that doesn't seem to render correctly regardless of how tabs stops are defined.

The pins labeled in the picture also don't match the tables that occur later in the file, specifically WRT Arduino Pins 10 through 13 (which are on PortE in the tables, and portC in the picture.  In fact, the picture has two different pins labeled "10~"

Here's a corrected picture (maybe.  Depends on whether the table or picture is correct!)
~~~
// ATMEGA4809 / ARDUINO
//
//                      SCL  SDA
//                     (A5*)(A4*) (7)  (2)                 (R)  (3~) (6~)
//                 PA4  PA3  PA2  PA1  PA0  GND  VDD  UPDI PF6  PF5  PF4  PF3
//
//                  48   47   46   45   44   43   42   41   40   39   38   37
//              + ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ +
//        PA5   1|                                                             |36  PF2
//        PA6   2|                                                             |35  PF1 (TOSC2)
//        PA7   3|                                                             |34  PF0 (TOSC1)
//   (9~) PB0   4|                                                             |33  PE3 (8)
//  (10~) PB1   5|                                                             |32  PE2 (13)
//   (5~) PB2   6|                                                             |31  PE1 (12)
//        PB3   7|                          48pin QFN                          |30  PE0 (11~)
//   (Tx) PB4   8|                                                             |29  GND
//   (Rx) PB5   9|                                                             |28  AVDD
//        PC0  10|                                                             |27  PD7 (VREF)
//        PC1  11|                                                             |26  PD6
//        PC2  12|                                                             |25  PD5 (A5)
//               + ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ ____ +
//                  13   14   15   16   17   18   19   20   21   22   23   24
//
//                  PC3  VDD  GND  PC4  PC5  PC6  PC7  PD0  PD1  PD2  PD3  PD4
//                                 (1)  (0)  (4)       (A0) (A1) (A2) (A3) (A4)

~~~

WestfW, 2018-10-31:
(I'm not sure what the "A5*" and "A4*" labels in the picture are...)

facchinm, 2018-10-31:
Thanks a lot for spotting this! A5 and A4 were "multiplexed" in the original version of the board with SDA/SCL to keep full compatibility with the UNO but we decided to drop it. Will update the image with the actual content immediately, thanks

facchinm, 2018-10-31:
Fixed by https://github.com/arduino/ArduinoCore-megaavr/commit/b93f57bf2ef4e78a8c9dc4cddc9bfec932b5f7ca

closed 2018-10-31 by facchinm