arduino/ArduinoCore-megaavr #6: Is NUM_TOTAL_PINS correct?

WestfW, 2018-11-07:
in variant/uno2018/pins_arduino.h there is:

```
#define NUM_DIGITAL_PINS            20
#define NUM_ANALOG_INPUTS           6
 :
#define NUM_TOTAL_PINS		(NUM_DIGITAL_PINS + NUM_ANALOG_INPUTS + NUM_TESTPOINT_PINS + NUM_OCCUPIED_PINS)
```
But the DIGITAL_PINS and ANALOG_INPUTS _overlap_, right?  Pin14 an Ain0 are the same.
(also, NUM_TOTAL_PINS turns out to be 39, but the `digital_pin_to_xxx[] `arrays have 41 elements.)

I think there are:
```
 20 normal digital and analog pins,
  2 dedicated pins for I2C
  3 dedicated pins for SPI on the ISP connector
  1 dedicated LED (hey - 6 randomish pins, just like NUM_ANALOG_INPUTS, so it comes out ALMOST right.) 
10 pins dedicated to the on-board IO (IMU, NINA, Debug USART),
  1 pin that doesn't seem to be connected anywhere (PC3)
  4 pins with dedicated functions TOSC1/2, VREF, RESET
```
Which adds up to the 41 IO pins that the datasheet mentions. (UPDI and 6 power pins totals 48.)

(It looks like the only place that NUM_TOTAL_PINS is used, currently, is in the digitalPintoXXX() macros, range checking.  So the extent of the actual "bug" is that Pin39 (VREF) and Pin40 (RESET) will not be usable for IO.  Which is not very important...)

**It would be nice to pin down exactly what these numbers are supposed to mean, since they appear to be part of the standard variant definitions.**

facchinm, 2018-11-08:
I agree, this is 
1- a clear bug in the core
2- a case of missing documentation

For the core I'm fixing the numbers right now; for the documentation, it is a broader problem. @SimonePDA how should we proceed?

closed 2018-11-08 by facchinm

per1234, 2018-11-08:
Regarding documentation, I think this would be the appropriate place:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#core-variants

I believe this falls under a previous issue report: https://github.com/arduino/Arduino/issues/4141