arduino/ArduinoCore-megaavr #8: Support for ATmega4809 Curiosity Nano

sstaub, 2018-11-09:
Can you add support for the ATmega4809 Curiosity Nano board from microchip?
[
https://www.microchip.com/DevelopmentTools/ProductDetails/DM320115](url)

facchinm, 2018-11-12:
Not sure about officially supporting it (since it's not an Arduino product and we usually don't add boards that are out of our control), but it will be definitely interesting to have it as a separate installable json :wink:

WestfW, 2018-11-24:
Xplained Pro, too?
I would have thought that Arduino's relationship with Atmel was close enough that it might be desirable to have Atmel Eval Board support for any eval board that uses the same CPU as an official Arduino.  At least for the boards that don't directly compete with (unfairly, in some cases) Arduino boards...

The 4809 board have significantly different SERCOM usage, which makes things a bit more complicated, I think.  Still looking.

goodchip, 2019-02-22:
I'm also very interressed by third party support of this board.

MCUdude, 2019-02-28:
For better 3rd party support you may try to use [https://github.com/MCUdude/MegaCoreX](https://github.com/MCUdude/MegaCoreX). It's still in a very early stage, but I welcome anyone to test it so improvements can be done

goodchip, 2019-03-01:
Nice job MCUdude, all my encouragements!

mbanzi, 2019-03-01:
@
[MCUdude](/MCUdude) wouldn't it be better to contribute to the existing core instead of creating a new one?

MCUdude, 2019-03-01:
Well, I will still contribute to this repo, but my main focus will be on my own core(s) and then maybe try to get a PR merged here afterwards. Here's why: my core/cores have a slightly different approach that the official Arduino boards. Mine are chip oriented and not board oriented, and give the users more hardware options in the tools menu. BOD and clocks are certainly important for more advanced users.

Let's concentrate on the MegaCoreX. My plan is to use these core files to support a variety of tinyAVR-0 and tinyAVR-1 chips too. I need to do some fundamental changes in order for this to happen. Changes like this doesn't really benefit users of the official Arduino boards, so PRs like this doesn't seem to be important for the devs to merge.

Here's also some important changes I'd need on order for me to use these core files directly:
* The ATmega3209/4809 need an alternative pinout to the Uno Wifi one. This is simply because the Uno Wifi pinout isn't very friendly when dealing with a chip instread of a board.
* Pinout for ATmega3208/4808, even though it does not exist any official hardware for these chips
* I want and need printf. I was planning to use ArduinoCore-API as a subtree, but I don't think I will anyway because it looks like I can't get it merged.
* Accurate timing needs to be generated by timer0 instead of timer3 in order to support smaller microcontrollers than 3209/4809. I'm waiting for facchinm to review this one
* Being able to use the pin swap functionality the new megaavr-0 series offers. I did it the ESP8266 way and added Serial.swap.

WestfW, 2019-03-02:
It would take a "corporate change of direction" to get the Arduino github repository to accept patches to either create or enhance new variants for "unsupported products" and/or from "competing" vendors.  (as facchinm  mentioned in his initial reply.)

It's a difficult problem, balancing the stability and internal expertise of a given core vs the customer (non-customer?) interest in other boards.

closed 2019-09-16 by facchinm