arduino/ArduinoCore-megaavr #11: Eliminate pgmspace?

WestfW, 2018-11-24:
This is just an idea, at the moment - I may get some statistics to support it (or not), soon.

The 0-seriese "megaavr" chips have a unified address space where flash is accessible the same as RAM, and so LPM instructions are no longer needed to access constant tables stored in flash.  An LDx from flash isn't actually any faster than an LPM, but there are a lot of extra addressing modes available
(indexing using X, Y, and Z, rather than only Z, for example), so I'd expect some overall improvement in behavior once the compiler gets it's register allocation optimization fingers in there.

That means that the various tables (notably the _digital_pin_to_* tables) can be allocated without specifying PROGMEM, and we don't need "pgm_read_*" functions to access them.

See also [https://www.avrfreaks.net/comment/2593311#comment-2593311](url)

facchinm, 2019-09-16:
To be merged with #53

facchinm, 2019-09-17:
Fixed by merging https://github.com/arduino/ArduinoCore-megaavr/commit/3b5508ec47aa72ecd9b50372cb0e213f3febe853

closed 2019-09-17 by facchinm