arduino/ArduinoCore-megaavr #20: Millis, micros and delay generated by timer3b

MCUdude, 2019-02-18:
Hi!

I'm working on a 3rd part Arduino core where the goal is to add proper support for the entire megaAVR-0 series (ATmega3208/4808, ATmega3209/4809) and hopefully a handful of new tinyAVRs in the future. I'm doing this based on this repo.

However an issue is that the 48kB devices (ATmega4808 and 4809) is the only devices out there with four 16-bit timers (type B). And, timer3b is used to generate micros in this core. This means that this core will only work with 48kB devices without heavy modification of the timer setup routines.

This may cause problems when it comes to code compatibility between different targets.

Why not use timer0b (TCB0) to generate millis, micros and delay? Timer0 have always beed used for this in the past, and all new AVRs (megaAVR-0, tinyAVR-0, tinyAVR-1 series) have TCB0 available. This would make porting these core files to other AVRs much simpler. 

I could of course change to TCB0 in my own repo, but it would be much better if TCB0 was the default timer no matter what hardware or Arduino core is used.

Everyone will benefit from Arduino support for the new AVR lineup 🙂
New users will discover the new and impressing lineup of AVR microcontrollers, and still being able to write programs for them in an API there already familiar with. And Microchip will sell more AVRs!

facchinm, 2019-02-19:
I totally appreciate the idea! TCB3 was chosen since the pins it can control as PWM output (PB5 and PC1) have a strong alternate function (SPI and DebugUART respectively).
Modifying the code to allow other timers via an #ifdef directive looks reasonable, and Tone can be used as a starting point (make the names more generic, tune the ISR).
What do you think about that?

MCUdude, 2019-02-19:
Tone is already using TCB1, which is fine IMO. Smaller ATtinys only has TCB0, so tone will not be available for these. However ATtinys with 16kB flash or more have TCB1, so these would work with tone out of the box. IMO it should be possible to [choose between TCB1, TCB2 and TCB3](https://github.com/arduino/ArduinoCore-megaavr/blob/dacdfc97579bb2800329e600fdb49d8fb0cb74a8/cores/arduino/Tone.cpp#L47-L51).

I suggest using TCB0 to generate micros/millis/delay without any ifdefs. Then the rest of the TCB* registers could be initialized in the variants.c file. 
However it would be great if the uno2018 variant supported the ATmega3209. It can be done if TCB3 initialization is left out if 3209 is detected as a target bu the compiler. Other AVRs (ATmega3208 for instance) will have its own variant files to match the hardware.

MCUdude, 2019-02-21:
Do you want me to submit a PR for this? I'm currently not sure how/if this will affect any PWM generation

facchinm, 2019-02-21:
A PR would be great :wink:

MCUdude, 2019-02-21:
Good! I'll do some testing first then

facchinm, 2019-03-11:
Should be solved by #24, reopen if needed

closed 2019-03-11 by facchinm

MCUdude, 2019-03-11:
So the ATmega4809 is not able to handle PWM and mills/micros with a single timer like the older AVRs can?

Kees-van-der-Oord, 2020-03-22:
Dear MCUDude,

I ran into the same challenge: implement timer functions on both the AVR and the MegaAVR chips. On the AVR chips, I use the TimerOne library, but that does not work on the MegaAVR chip. I wrote a timer library for the TCB timer that that has the same interface as the Timer1 object. This allows you write code that works on both the AVR and MegaAVR architectures.
The library is here:
[https://github.com/Kees-van-der-Oord/Arduino-Nano-Every-Timer-Controller-B
](https://github.com/Kees-van-der-Oord/Arduino-Nano-Every-Timer-Controller-B)
I hope this helps you !
WR,
Kees

facchinm, 2020-03-23:
@Kees-van-der-Oord if you rearrange the library a bit based on https://arduino.github.io/arduino-cli/library-specification/ it would be cool to include it in the library manager (if you wish :wink: )