arduino/ArduinoCore-megaavr #27: TimeLib.h updated for use RTC circuit embedded with megaavr

goodchip, 2019-03-08:
Just for information,

I've updated original TimeLib.h library for use with internal RTC clock of Arduino Uno WiFi rev 2 
https://github.com/goodchip/Time

Examples here: https://github.com/goodchip/Time/tree/master/examples/TimeInternalRTC

**Features:**

- use internal RTC circuit for real clock in board (not based on millis() function)
- additional attach/detach interrupt function to run user function at 1Hz synchronized by RTC
- 1/4096s top syncho between RTC and real second when setTime is called
- 100% compatible with 328p code (run without RTC) if interrupt are not used

**Current limitation:**

- use only external quartz for the moment.
- alpha version

_**Have fun!**_

Sylvain

goodchip, 2019-03-20:
**New features:**
- adjustable IRQ call frequency added (from 1s to 1/8192s)
- adjustable synchro precision RTC added (from 1s to 1/8192s)
- new maths routines very optimized

goodchip, 2019-04-10:
**Major upgrades - very stable and usefull version**

New! 2 interrupts features:
- InternalRTC.attachInterrupt: attach a IRQ user function called at the freq frequency
- InternalRTC.attachClockInterrupt: attach a IRQ user function called each new second

Great improvements:
- removal of the function setPrecision(), now useless
- now always maximal RTC cycles precision without cpu load

MCUdude, 2019-04-11:
I think I've figured out why the sketch didn't run on my UNO Wifi Rev2. It seems like one of the capacitors (C904 in the schematic) connected to the 32.768kHz crystal is swapped out with a resistor instead. I measured mine to 19.1k. Looking a bit more in the schematic it turns out that 19.1k is the value of one of the feedback resistors for the 5V buck converter. I'm pretty sure this is caused by the PCBA pick and place machine is incorrectly set up. I'm also pretty sure many (all?) Uno Wifi Rev2 boards from this exact batch have this issue. How about your board @per1234? Does your board have the same flaw? Who do we report this to?

I do have a Curiosity Nano, but I've cut the UPDI line because I was doing some testing with an external programmer. I accidentally ripped off a pad, so I need to use a microscope to solder it back on again.

The black component inside the red circle is supposed to be brown.
[image]

per1234, 2019-04-11:
>How about your board @per1234?

@MCUdude that component on my board looks the same as yours.

>Who do we report this to?

I think your comment here will serve to get someone knowledgeable on this subject to take a look at the situation. Speaking more generally, I think the arduino/Arduino issue tracker is the best place to report hardware issues with official Arduino boards, since it acts as a catch-all for issues that don't match with any of Arduino's other public issue trackers. I don't think it will be necessary to open an issue there in this particular case though, since this comment will act as enough of a report.

Thanks!

goodchip, 2019-04-12:
@MCUdude 

I understand better why my fork doens't run correctly on your board :(

Visibly, I've not this problem with the batch of my board.
[image]

Sorry for you, too bad there is probably quite expensive postage because I have everything you need to fix this.

MCUdude, 2019-04-12:
Don't worry! I should be able to get my hands on a 0402 ~10pF capacitor soon

goodchip, 2019-05-14:
@MCUdude : Do you have somes news about your card? :)

MCUdude, 2019-05-14:
Sorry, I've been unavailable for a week. Been on a business trip to China, så no Gmail notifications. I bought myself a nice microscope there, so I'll borrow the capacitor from my xplained board. I'm pretty jet lagged right now, so give me a couple of days. This is also a nice test application for the microscope 😉

PickyBiker, 2019-05-18:
I think my board may have the same problem. I just ordered it from Amazon. The parts next to the xtal are black on the bottom and brown on the top. I too was able to measure the black component at 19.1k. 
[image]

I don't think I can replace that part myself. I guess I will have to return it to Amazon. 

Before I do that, is there any doubt about this being wrong? Could the 19.1k resistance be coming from other circutry or inside the xtal itself?

MCUdude, 2019-05-19:
I was able to borrow the tiny 0402 capacitors from a curiosity nano board, and I manage to solder them to the UNO Wifi Rev2 board. The library is now working like a charm!

> Could the 19.1k resistance be coming from other circuitry or inside the xtal itself? Are we absolutely sure that is the C904 component?

I know the value seems odd. this resistor is used as a voltage divider for the 5V switching regulator. I'm 100% sure that this is a production fault. I'm also pretty confident that C904 is the component. I looked through the schematic and compared it with the Gerber files to be sure.

PickyBiker, 2019-05-19:
If I return this to Amazon there is a good chance the replacement will have the same problem. I am 71 years old and I don't have great vision and am not too steady of hand. There is a low probability that I could successfully change that tiny part. If there is anyone monitoring this site from Arduino, is there any way to get a replacement for this board?

MCUdude, 2019-05-19:
> If I return this to Amazon there is a good chance the replacement will have the same problem.

It's likely, yes. You can buy a 4809 Curiosity Nano instead though. This also has an onboard 32.768 kHz crystal and works with Arduino IDE if you use [MegaCoreX](https://github.com/MCUdude/MegaCoreX).

PickyBiker, 2019-05-19:
MCUDude, thanks for the compliment.  The Curiosity Nano doesn't have WiFi nor does it support the Arduino shields. Also, I am of the age where I think Arduino Company should take some corrective action on an obvious manufacturing error. Still, if I had confidence that I could buy another WiFi Rev2 that didn't have this problem, I would do it. It's not about the money, it's about expecting a company to deliver what it advertises. I am a big fan of Arduino's and they keep me busy having fun in retirement.

per1234, 2019-05-19:
>is there any way to get a replacement for this board?

Please contact Arduino Store support here:
https://store.arduino.cc/store-support

PickyBiker, 2019-05-19:
I submitted a request for replacement via that site. Thank you.

PickyBiker, 2019-05-23:
I just sent the following message  to Arduino's customer Support agent.

"Hello, Sravya,

On May 19th I sent a request to return a defective UNO WiFi Rev2. You said I had to return it to the distributor where I bough it so I returned it to Amazon.

I then I purchased one from your online store and that board arrived today. Unfortunately, it has the same manufacturing defect that I pointed out to you with the first board. The problem is component C904 is supposed to be a 20pf capacitor but there is a 19.1K resistor installed there instead. That manufacturing defect causes the RTC crystal to fail to operate so the RTC does not work. 

I suspect there are many built incorrectly like this because a pick and place machine that places the components on the board is setup wrong and until someone fixes that machine's setup it will continue to manufacture all of them incorrectly. There could be hundreds or thousands of these boards that are built incorrectly. 

Please confirm for me that Arduino is aware of this manufacturing problem and they are taking steps to correct it.  
Secondly please arrange for a return now with a refund. I do not want a replacement until after the problem is confirmed to be corrected and your stock has been replaced or repaired. 

I await your reply.
Thank you,
Mike"

PickyBiker, 2019-05-24:
Just got confirmation the problem is being corrected.

F-Marion, 2019-07-21:
I have requested a replacement two times now and both replacement boards have the same defect.