arduino/ArduinoCore-megaavr #28: Warning when second argument in digitalWrite is not PinStatus

MCUdude, 2019-03-11:
It seems like the second argument of digitalWrite is expecting a constant from `PinStatus`. However this isn't always ideal, and we get an ugly warning like shown below when this is not the case.

```
/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_206750/BlinkWithoutDelay.ino: In function 'void loop()':
/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_206750/BlinkWithoutDelay.ino:69:34: warning: invalid conversion from 'int' to 'PinStatus' [-fpermissive]
     digitalWrite(ledPin, ledState);
                                  ^
In file included from /Users/hans/Documents/Arduino/hardware/MegaCoreX/avr/cores/coreX-corefiles/api/ArduinoAPI.h:52:0,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/avr/cores/coreX-corefiles/Arduino.h:23,
                 from /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_868899/sketch/BlinkWithoutDelay.ino.cpp:1:
/Users/hans/Documents/Arduino/hardware/MegaCoreX/avr/cores/coreX-corefiles/api/Common.h:104:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)'
 void digitalWrite(pin_size_t pinNumber, PinStatus status);
      ^
```

``` c
// Blink without Delay

// constants won't change. Used here to set a pin number:
const int ledPin =  LED_BUILTIN;// the number of the LED pin

// Variables will change:
int ledState = LOW;             // ledState used to set the LED

// Generally, you should use "unsigned long" for variables that hold time
// The value will quickly become too large for an int to store
unsigned long previousMillis = 0;        // will store last time LED was updated

// constants won't change:
const long interval = 1000;           // interval at which to blink (milliseconds)

void setup() {
  // set the digital pin as output:
  pinMode(ledPin, OUTPUT);
}

void loop() {
  // here is where you'd put code that needs to be running all the time.

  // check to see if it's time to blink the LED; that is, if the difference
  // between the current time and last time you blinked the LED is bigger than
  // the interval at which you want to blink the LED.
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you blinked the LED
    previousMillis = currentMillis;

    // if the LED is off turn it on and vice-versa:
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }

    // set the LED with the ledState of the variable:
    digitalWrite(ledPin, ledState);
  }
}
```

MCUdude, 2019-04-20:
Can we add a simple template wrapper around digitalWrite to automatically cast whatever datatype is passed to `pinStatus`? Or do you think this is an ugly solution @facchinm ?

per1234, 2019-04-20:
There is a proposal that should resolve the issue here:
https://github.com/arduino/ArduinoCore-API/issues/25#issuecomment-461327599

MCUdude, 2019-04-20:
@per1234 I totally missed this one. Thanks for linking it in. I'll leave my comment there instead.