arduino/ArduinoCore-megaavr #32: Arduino Wifi Rev2, mqtt, wifi + servo not compatible?

arduino-cyberspace, 2019-04-18:
Hi,
I have no idea if this is the right place to post this issue. If wrong, just delete it, my apologies.

I have posted it on arduino.cc and haven't got one single suggestion or answer at all. I don't want to post the same info in two places so I basically will just post the link. 

**But here comes a short description:**
The issue is that I cannot get Arduino Wifi Rev2 to work with mqtt, wifi and servo at the same time. mqtt and wifi works fine, servo works fine but as soon as I add those together everything gets very unstable. No matter what pin I use for the servo (which was my original problem but then I found out which one to use (8)). And it is also very unstable without the servo even connected, that is when I realized I might have a real problem somewhere. 

I have tried to boil the problem down to a simple one and now I am down to one simple sketch (an example sketch) with two extra lines of code. As soon as I add the servo everything becomes unstable. And I have no idea why.

Here is the link to what I think is a clear definition of the problem. Would very much appreciate any form of tips or suggestion on how to go on because I am very close to give up totally after a long time of trying to find out what is wrong. Maybe I have totally missed something.

https://forum.arduino.cc/index.php?topic=610037.0

facchinm, 2019-04-19:
Hi @arduino-cyberspace ,
I have no clear idea of what the problem can be, but I can share with you a sketch of the demo we showcased at Maker Faire Rome with UNO WiFi rev2 controlling Braccio (6 servos) via WiFi commands (no mqtt but should be similar).

Are you powering the servos from the board itself or externally? Voltage spikes due to large current absorption could be the cause of the wifi malfunction, but also on the code side there could be very thigh loops that the SPI bus cannot easily handle. If you want to past a minimal example triggering the issue I can take a look.

arduino-cyberspace, 2019-04-19:
Hi,
thank you for coming back to me. As I wrote in my description, the servo is not even connected to the board so your idea (though good and something I thought of also) is not applicable I think since the servo is not connected at all.

The example I attached on the other site that I linked to is now attached here also:
[WfiSimpleReceive (1).txt](https://github.com/arduino/ArduinoCore-megaavr/files/3098589/WfiSimpleReceive.1.txt)

If you would have a look and maybe come back with some idea, it would be great. Especially with the thing you write about "tight loops" - that makes sense to me - should I just add a Delay or something you mean or how can I make them less "tight"..?

Best regards
Sam

facchinm, 2019-04-23:
Got it. Probably the timer ISR controlling the servos is a bit too long and tends to break the wifi communication (which happens via ISR too). I'm trying it "live" as soon as possible.
Thanks for the sample code!

arduino-cyberspace, 2019-04-23:
Hi, 
what I have managed to figure out so far is (since you mentioned something I did not think about (tight loops)) that if I put in a delay(1000), then it is much more stable - the first step to get it to work. Still nothing connected, just the board itself. But should it really be so unstable without this delay(1000)? I have tried many different delays but a smaller number than 1000 still makes it drop the mqtt sometimes. With 1000 it is very rare, it basically works all the time.

egilkv, 2019-08-04:
Since your program seems to involve quite a bit of transmission over Serial, my educated guess would be that the problem you have is a very serious bug in the current Uno Wifi Rev2 library. 

I have filed a bug for it here: https://github.com/arduino/ArduinoCore-megaavr/issues/43 and a resolution here; https://github.com/arduino/ArduinoCore-megaavr/pull/42

I am hoping this fix will be accepted. but if not, you can always try my fork at https://github.com/egilkv/ArduinoCore-megaavr

arduino-cyberspace, 2019-08-08:
Hi,
what you write does not surprise me I am afraid. That is the general feeling I have had so far with this new board, even the specs (OTA updates) changed after the release of it. Thank you very much for looking in to this, I think I will demount my Arduino Wifi Rev2 from where it is installed (in the garage) in the near future, change the software to your fork and then see what happens.

facchinm, 2019-09-18:
@arduino-cyberspace it looks like #42 solved a lot of problems with sketches heavily using Serial. May you test core 1.8.5 and close the issue if it's fixed? Thanks!

arduino-cyberspace, 2019-09-18:
Hello @facchinm,
I have tested 1.8.5 and for me it is the same result as 1.8.3. If I check the mqtt-connection once every 500:th loop that Arduino does then it works fairly well. If I do it once every single loop or once every 20.000 loop - it breaks the connection to mqtt very often (more than once a minute). 

So I cannot say that this is solved from _my_ point of view but maybe I have something else wrong in the code that breaks it. But I have spent a lot of time trying to solve this and I have no more ideas.
I think however that the problem is less frequent with 1.8.3 or later, my feeling was that it got better after the fix. So maybe we are talking two different problems here, I do not know.

This is what I do once every 500 time Arduino loops, maybe this is wrong as well.

`if (!mqttClient.connected()) {
    Serial.println("MQTT connection lost - trying again.");
    connectMQTT();
  }

mqttClient.loop();
`

If you feel that the problem caused by [https://github.com/arduino/ArduinoCore-megaavr/pull/42](url) is resolved for other users then you can close this - I can only say that I still have some instability issue.

Thank you for your time!