arduino/ArduinoCore-megaavr #33: Problem Downloading code to Wifi Rev2

TimEllis2019, 2019-04-29:
My code compiles fine but when I get to the downloading part of a new install I get the following error.  At the bottom of the error list it says the board is not connected but when I get the board info the window shows the correct info.

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\telli\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\telli\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\telli\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10809 -build-path C:\Users\telli\AppData\Local\Temp\arduino_build_873433 -warnings=none -build-cache C:\Users\telli\AppData\Local\Temp\arduino_cache_444121 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -verbose C:\Users\telli\Documents\Arduino\temp_control\temp_control.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\telli\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\telli\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\telli\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10809 -build-path C:\Users\telli\AppData\Local\Temp\arduino_build_873433 -warnings=none -build-cache C:\Users\telli\AppData\Local\Temp\arduino_cache_444121 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -verbose C:\Users\telli\Documents\Arduino\temp_control\temp_control.ino
Using board 'uno2018' from platform in folder: C:\Users\telli\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.6.26
Using core 'arduino' from platform in folder: C:\Users\telli\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.6.26
Detecting libraries used...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino/api/deprecated" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\variants\\uno2018" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\sketch\\temp_control.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino/api/deprecated" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\variants\\uno2018" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\sketch\\temp_control.ino.cpp" -o "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino/api/deprecated" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\variants\\uno2018" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\sketch\\temp_control.ino.cpp" -o "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\sketch\\temp_control.ino.cpp.o"
Compiling libraries...
Compiling core...
Using previously compiled file: C:\Users\telli\AppData\Local\Temp\arduino_build_873433\core\variant.c.o
Using precompiled core: C:\Users\telli\AppData\Local\Temp\arduino_cache_444121\core\core_arduino_megaavr_uno2018_mode_off_dcb64d4f6a6de1966fae6a8c32168afe.a
Linking everything together...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega4809 -o "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433/temp_control.ino.elf" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\sketch\\temp_control.ino.cpp.o" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433\\core\\variant.c.o" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433/..\\arduino_cache_444121\\core\\core_arduino_megaavr_uno2018_mode_off_dcb64d4f6a6de1966fae6a8c32168afe.a" "-LC:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_873433" -lm
[... cut ...]
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

btabuenca, 2019-05-07:
After some months working with the board, suddenly I got the same message:

------------------------------------
```
Arduino: 1.8.9 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

Sketch uses 1534 bytes (3%) of program storage space. Maximum is 49152 bytes.
Global variables use 22 bytes (0%) of dynamic memory, leaving 6122 bytes for local variables. Maximum is 6144 bytes.
avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67
avrdude: error reading signature data, rc=-67

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

TimEllis2019, 2019-05-07:
Bernardo,

Did you change your code so that you were forcing the compiler to use more flash instead of RAM?  IE 

 

Serial.print(F("Write something on the Serial Monitor that is stored in FLASH"));

 

Or

const char StyleHTML[] =" some long string like a web page”;

 

My issue started after I started using these.

-tim

btabuenca, 2019-05-07:
Unfortunately, that is not my case.

btabuenca, 2019-05-07:
Just tried with a 2nd Arduino UNO wifi rev-2 that was working this morning. Now I get the same message and did not modified any line of code. 

This is how things happened:

1. I tested it in Genuino linux this morning. Worked!
2. I tested it in Genuino Mac this afternoon. Worked!
3. I tested it in Genuino linux this afternoon. Did not worked :(
4. I tested it in Genuino Mac this afternoon. Did not worked :(

Now I have 2 Arduino UNO wifi rev-2 not working. I cannot upload any program

Please let me know if I could help

facchinm, 2019-05-08:
Hi everyone,
we are bisecting the issue to understand what could be the root cause. Feel free to contact the support for a replacement in the meantime.

agdl, 2019-05-08:
@TimEllis2019 Can you please send us a top photo of your board?

btabuenca, 2019-05-08:
Here you are a photo from mine

[image]

TimEllis2019, 2019-05-08:
Photo attached.  Please let me know if you need anything else.

-tim

TimEllis2019, 2019-05-08:
Martino, would it be helpful to send my board in?

per1234, 2019-05-08:
>Photo attached.

@TimEllis2019 email attachments don't work on GitHub. Please open this issue thread in your browser: https://github.com/arduino/ArduinoCore-megaavr/issues/33
and then attach the photo in a reply made right to the thread on GitHub, rather than via an email reply.

TimEllis2019, 2019-05-08:
Let me know if you want me to send back my board if that is of use.

-tim

TimEllis2019, 2019-05-09:
I don’t see a picture.  Gethub sent my email back saying no attachments.  Then failed because my email was too big.

 

 <https://user-images.githubusercontent.com/895658/57411914-3c0b3280-71ef-11e9-915c-03c3ba6b1bf1.jpg>

facchinm, 2019-05-13:
@TimEllis2019 , sure, feel free to contact support linking this thread so we'll be able to track the issue is a sane way.

jguen, 2019-05-14:
Since yesterday I have the same problem with my Arduino Uno Wifi Rev 2. It would be great if someone finds a solution for this bug. I also wrote an email to the Arduino support and I am waiting for an answer. 

Is it maybe possible to exchange the defect Arduino, because it is in the warranty time? 

My actual project is a multi medial steering platform for the Tinkerkit Braccio and the Arduino is the leading component for me.

TimEllis2019, 2019-05-16:
[image]
Sorry my picture did not go thru.

OmriEg, 2019-05-22:
I also suffer from the same issue. i couldn't download the code to Arduino wifi rev 2. 
error massage:
Sketch uses 14242 bytes (28%) of program storage space. Maximum is 49152 bytes.
Global variables use 1027 bytes (16%) of dynamic memory, leaving 5117 bytes for local variables. Maximum is 6144 bytes.
avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67
avrdude: error reading signature data, rc=-67

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

uny update on this issue?

jan1za, 2019-05-22:
I am getting the same issue to. It was working perfectly. Ran the WifiNina check firmware version sketch, it worked. Loaded the WifiNina update firmware sketch, tried to update and got this:

Sketch uses 6806 bytes (13%) of program storage space. Maximum is 49152 bytes.
Global variables use 1397 bytes (22%) of dynamic memory, leaving 4747 bytes for local variables. Maximum is 6144 bytes.
avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67
avrdude: error reading signature data, rc=-67

jan1za, 2019-05-22:
Sketch uses 1544 bytes (3%) of program storage space. Maximum is 49152 bytes.
Global variables use 22 bytes (0%) of dynamic memory, leaving 6122 bytes for local variables. Maximum is 6144 bytes.
C:\Users\nicho\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\nicho\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -b115200 -e -D -Uflash:w:C:\Users\nicho\AppData\Local\Temp\arduino_build_561657/Blink.ino.hex:i -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\nicho\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : E870C4BC24F2C7F711C8
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

An error occurred while uploading the sketch
Reading | avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67

avrdude done.  Thank you.

facchinm, 2019-05-24:
Didn't want to close the issue directly, but thanks to the replacements you sent us we have a solution :slightly_smiling_face: 

First of all, the problem is probably due to a bug in avrdude, which incorrectly writes the CRC bits (fuse SYSCFG0 - 0x05).
Once CRC is wrong, the only way to unlock the chip is through an `unlock/chip erase` procedure which differs from the normal chip erase.
This procedure was triggered only in case the return value of `avr_signature` was `-68`.
Due to other changes, mEDBG reports `-67` as errors, so it doesn't trigger the right erase procedure.

I'm currently producing new `avrdude` builds that should fix the error. In case you want to test before the official release, the right commandline is 
```
bin/avrdude -C etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -b115200 -e -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m
```

facchinm, 2019-05-24:
Core 1.8.1 had just been released, boards exibiting the issue will just self-fix on the first upload using the new core :smile: 
@jan1za  @TimEllis2019  @per1234 @btabuenca @jguen @OmriEg

closed 2019-05-24 by facchinm

TimEllis2019, 2019-05-24:
I just downloaded the new core and was able to reprogram my “failing Wifi Rev 2” board.

 

One comment, when I went to the board manager it did not tell me a new version of the code was available like I remember in the past when changes were made. I had to look at the download list to see 1.8.1.

 

Thanks for your efforts on this.

jan1za, 2019-05-24:
Thank you for all the work. Unfortunately I exchanged the WiFi rev 2 for a
yun as I  needed a working demo for today else I would have tried the fix.

Thank you once again for all your efforts.

btabuenca, 2019-05-24:
Thanks for your work guys! Now it works.

Just for your information in case it helps ... now I get the following warning:

```
avrdude: WARNING: invalid value for unused bits in fuse "fuse5", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xcd instead of 0xc9 (double check with your datasheet first).
```

EbStewart, 2020-01-04:
> Just for your information in case it helps ... now I get the following warning:
>
> ```
> avrdude: WARNING: invalid value for unused bits in fuse "fuse5", should be set to 1 according to datasheet
> This [...]

I get this too. Using an Uno wifi rev2 in both arduino IDE and Visual Studio 2017 with vMicro plugin.

Things seem to work anyway, do I need to fix anything? if so, how can I 'use 0xcd instead of 0xc9'?

per1234, 2020-01-04:
>do I need to fix anything?

@EbStewart no. See https://github.com/arduino/Arduino/issues/9443

TimEllis2019, 2020-01-04:
My version of the compiler works fine.  I get the same warnings but my code works.

-tim

 

@EbStewart <https://github.com/EbStewart>  no. See arduino/Arduino#9443 <https://github.com/arduino/Arduino/issues/9443>

EbStewart, 2020-01-04:
Mine seems to as well, but I want it to continue to do so. If this blows up at the last minute I'm screwed, so I want to be sure.

EbStewart, 2020-01-04:
@per1234 Thank you.

barrypress, 2020-07-26:
Should this still be happening as of late July?

antaxiom, 2020-11-14:
2 months since the last post made here, I still have this issue and I'm unable to fix it. I thought it was a linux issue (I used arch linux for the record) and now I've tried it on windows and I have word for word the same issue. I've tried many versions of the megaARV boards including the latest (1.8.6). I have an equivalent of the UNO WiFi Rev 2 but it seems to be identical to my understanding. If someone could shed some more light on this issue that would be great, please and thank you.

manilal2, 2021-01-26:
I have custom made PCB using ATMEGA4809 (according to schematic ( https://camo.githubusercontent.com/0318ea09473584a26d3f653e1501e8ba43a524259eb7dfe0a36920a9d6d7bd05/68747470733a2f2f692e696d6775722e636f6d2f784e6778627a582e706e67))

I am getting following error, I am using atmel ICE (updi) 

 
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/root/.arduino15/packages/MegaCoreX/hardware/megaavr/1.0.7/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : atmelice_updi
avrdude: usbdev_open(): Found Atmel-ICE CMSIS-DAP, serno: J42700031106
avrdude: max packet size expected 912, but found 512 due to EP 0x01's wMaxPacketSize
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel-ICE (ARM/AVR) in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.41 (rel. 137)
         Serial number   : J42700031106
         Vtarget         : 5.01 V
         JTAG clock megaAVR/program: 1000 kHz
         JTAG clock megaAVR/debug:   1000 kHz
         JTAG clock Xmega: 1000 kHz
         PDI clock Xmega : 1000 kHz

avrdude: Partial Family_ID returned: ""
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-70
avrdude: error reading signature data, rc=-70

avrdude done.  Thank you.

Error while burning bootloader.

I am trying to burn bootloader ( on UART1, I have connections according to  my schematics)

TimEllis2019, 2021-01-27:
Hi, this is Tim Ellis’ wife, Debbie Ellis.

 

Tim Ellis died unexpectedly on January 7.  Can you please remove him from your lists, so that no more emails will come to his computer?

 

Thank you, Tim’s wife

per1234, 2021-01-27:
My condolences Debbie.

Arduino doesn't have any way to unsubscribe your account. If you click the "unsubscribe" link at the bottom of the email it will unsubscribe you from this specific thread, but that doesn't affect the other threads you are subscribed to. You can control the email notification preferences for the entire account here:
https://github.com/settings/notifications

manilal2, 2021-01-27:
how do you execute chip erase using updi...?

manilal2, 2021-01-29:
I was able to program third board using atmel ( microchip) studio , where I erased it before programming , after programming verification was successful , when I try to flash sketch ( led blink sketch ) it gave me (bootloader and arduino IDE verification error , subsequently I tried to reprogram atmega4809 ( 48 pin)  bootloader and I was not able to put in program mode , is there some fuse setting which will allow me to put atmeag4809 in program mode , right now I am unable to program it ( it says unable to put atmega4809 in programming mode , once program I am unable to erase it )

also what bootloader I should be using my parameters are 20Mhz internal clock, baud rate 115200,optiboot UART on serial 1 ,BOD 2.6v ,48 pin standard, Reset pin Reset , can some one give me filename for bootloader and where it is in arduino ide...?