arduino/ArduinoCore-megaavr #36: HardwareSerial.h is missing

sstaub, 2019-06-18:
Using the OSC library from CNMAT, I get an error when compiling, that HardwareSerial is missing.

Arduino: 1.8.9 (Mac OS X), Board: "Arduino Nano Every, None (ATMEGA4809)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/sstaub/Library/Arduino15/packages -hardware /Users/sstaub/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/sstaub/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/sstaub/Documents/Arduino/libraries -fqbn=arduino:megaavr:nona4809:mode=off -vid-pid=0X2341_0X0058 -ide-version=10809 -build-path /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_build_59850 -warnings=default -build-cache /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_cache_949057 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16 -prefs=runtime.tools.avrdude-6.3.0-arduino16.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16 -verbose /Users/sstaub/Documents/Arduino/lighthack_box_2b_p/lighthack_box_2b_p.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/sstaub/Library/Arduino15/packages -hardware /Users/sstaub/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/sstaub/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/sstaub/Documents/Arduino/libraries -fqbn=arduino:megaavr:nona4809:mode=off -vid-pid=0X2341_0X0058 -ide-version=10809 -build-path /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_build_59850 -warnings=default -build-cache /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_cache_949057 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16 -prefs=runtime.tools.avrdude-6.3.0-arduino16.path=/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16 -verbose /Users/sstaub/Documents/Arduino/lighthack_box_2b_p/lighthack_box_2b_p.ino
Using board 'nona4809' from platform in folder: /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1
Using core 'arduino' from platform in folder: /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1
Build options changed, rebuilding all
Detecting libraries used...
/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/deprecated -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/variants/nona4809 /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_build_59850/sketch/lighthack_box_2b_p.ino.cpp -o /dev/null
/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/deprecated -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/variants/nona4809 -I/Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal/src /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_build_59850/sketch/lighthack_box_2b_p.ino.cpp -o /dev/null
/Users/sstaub/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/deprecated -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino -I/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/variants/nona4809 -I/Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal/src -I/Users/sstaub/Documents/Arduino/libraries/OSC /var/folders/x6/n06cf3c50g3ctgcw5mjwcrjw0000gn/T/arduino_build_59850/sketch/lighthack_box_2b_p.ino.cpp -o /dev/null
In file included from /Users/sstaub/Documents/Arduino/lighthack_box_2b_p/lighthack_box_2b_p.ino:34:0:
/Users/sstaub/Documents/Arduino/libraries/OSC/SLIPEncodedSerial.h:11:10: fatal error: HardwareSerial.h: No such file or directory
 #include <HardwareSerial.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Using library LiquidCrystal at version 1.0.7 in folder: /Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal 
Using library OSC at version 1.3.5 in folder: /Users/sstaub/Documents/Arduino/libraries/OSC 
exit status 1
Error compiling for board Arduino Nano Every.

facchinm, 2019-06-18:
Hi @sstaub ,
`HardwareSerial.h` is now available under `api` subfolder (for every project based on [ArduinoCore-api](https://github.com/arduino/ArduinoCore-API)) for coherency.
To solve the problem with OSC, please apply this patch
```patch
diff --git a/SLIPEncodedSerial.h b/SLIPEncodedSerial.h
index ce733be..1198966 100644
--- a/SLIPEncodedSerial.h
+++ b/SLIPEncodedSerial.h
@@ -8,8 +8,11 @@ Extends the Serial class to encode SLIP over serial
 
 #include "Arduino.h"
 #include <Stream.h>
+#ifdef ARDUINO_API_VERSION
+#include <api/HardwareSerial.h>
+#else
 #include <HardwareSerial.h>
-
+#endif
 
 class SLIPEncodedSerial: public Stream{
```

sstaub, 2019-06-18:
The OSC problem is solved with this patch.

closed 2019-06-18 by sstaub