arduino/ArduinoCore-megaavr #37: Serial output is garbled if done immediately after calling Serial.begin()

per1234, 2019-06-24:
Arduino Uno WiFi Rev2
Arduino megaAVR Boards 1.8.1
Arduino IDE 1.8.10 Hourly Build 2019/05/21 09:33 / Arduino IDE 1.9.0
Windows 10 64 bit / Ubuntu 19.04-beta build 107

The following code:
```c++
void setup() {
  delay(1000);  // to prove the issue is not related to printing immediately after startup
  Serial.begin(9600);
  delayMicroseconds(100);
  Serial.println("hello");  // called after a delay, so it prints fine
  Serial.println("hello");  // just to be sure everything is still printing OK
  Serial.begin(9600);
  Serial.println("hello");  // called without a delay, so it's garbled
}
void loop() {}
```
results in this output:
```
hello
hello
⸮⸮⸮⸮5
```
Without a delay of around 100 us between the call to `Serial.begin()` and `Serial.println()`, the output is garbled. As you can see from the output, this is not related to being soon after the program starts. It occurs even with subsequent calls to `Serial.begin()` (it also would have after the first call without the delay).

I've never experienced this with any other official Arduino board.

I've seen at least two threads on the Arduino Forum so far where this caused confusion.

mraardvark, 2019-06-24:
If the second "hello" is still being sent when you reconfigure the UART, I would not be surprised with garbage.  Perhaps flush first, or add significant delay before the second .begin() to be sure its done sending and safely in idle?

per1234, 2019-06-24:
The issue still occurs even with a call to `Serial.flush()` and/or long delay before the second `Serial.begin()`.

But it's a good criticism of my demo code. It's funny because I originally had a `delay(1000)` there, but for some reason I decided it didn't add anything of value to the demo code and removed it at the last minute before submitting the issue.

mraardvark, 2019-06-24:
So does the same behaviour happen if you call Serial.end() inbetween?

per1234, 2019-06-24:
Yes it does.

I think I made a bad decision by using two calls to `Serial.begin()` in my demo code because it seems to be just muddying things. Here's the bare minimum demo of the issue:
```c++
void setup() {
  delay(1000);  // to prove the issue is not related to printing immediately after startup
  Serial.begin(9600);
  Serial.println("hello");
}
void loop() {}
```
output:
```
⸮⸮⸮⸮5
```

facchinm, 2019-06-25:
Hi @per1234 ,
the problem seems related with a spike on TX pin that confuses the USB to serial converter.
Moving pin initialization later seems to fix the issue on my side. I'm attaching the patch
```patch
diff --git a/cores/arduino/UART.cpp b/cores/arduino/UART.cpp
index fa8098c..8e0520b 100644
--- a/cores/arduino/UART.cpp
+++ b/cores/arduino/UART.cpp
@@ -147,13 +147,6 @@ void UartClass::begin(unsigned long baud, uint16_t config)
 
     _written = false;
 
-    //Set up the rx pin
-    pinMode(_hwserial_rx_pin, INPUT_PULLUP);
-
-    //Set up the tx pin
-    digitalWrite(_hwserial_tx_pin, HIGH);
-    pinMode(_hwserial_tx_pin, OUTPUT);
-
     int8_t sigrow_val = SIGROW.OSC16ERR5V;
     baud_setting *= (1024 + sigrow_val);
     baud_setting /= (1024 - abs(sigrow_val));
@@ -169,6 +162,13 @@ void UartClass::begin(unsigned long baud, uint16_t config)
 
     (*_hwserial_module).CTRLA |= USART_RXCIE_bm;
 
+    //Set up the rx pin
+    pinMode(_hwserial_rx_pin, INPUT_PULLUP);
+
+    //Set up the tx pin
+    digitalWrite(_hwserial_tx_pin, HIGH);
+    pinMode(_hwserial_tx_pin, OUTPUT);
+
     // Restore SREG content
     SREG = oldSREG;
 }
```
If you could test it in your setup (also to make sure it doesn't regress anything else) I'll be more than happy to merge it.

per1234, 2019-06-25:
Thanks @facchinm! That patch fixes the problem for me also. I ran some other random sketches that use Serial without encountering any problems, but didn't do any more formal testing than that. I'll leave the patch in place and report if I encounter any problems.

commit ca37340f (2019-06-28): Move UART pins initialization later

Fixes #37
closed 2019-06-28 by facchinm