arduino/ArduinoCore-megaavr #47: Installing Arduino megaAVR Boards breaks ATTinyCore ISP sketch uploads

JChristensen, 2019-08-07:
Steps to reproduce:
1. Install Arduino 1.8.9.
2. Install [ATTinyCore 1.2.5](https://github.com/SpenceKonde/ATTinyCore) using the [Boards Manager Installation instructions on GitHub](https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md).
3. Using the following settings on the Arduino Tools menu and an Adafruit USBtinyISP programmer, upload a sketch to an ATtiny85:
```
Board: ATtiny25/45/85
Chip: ATtiny85
Clock: 1MHz (internal)
B.O.D. Level: B.O.D. Disabled
Save EEPROM: EEPROM retained
Timer 1 Clock: CPU
LTO: Disabled
Programmer: USBtinyISP (ATTinyCore)
```
4. Upload succeeds, see attached verbose upload output (upload-ok.txt, attached)
5. Using the Boards Manager, install Arduino megaAVR Boards by Arduino version 1.8.3.
6. Retry step (3) above; upload fails, see attached output (upload-err.txt, attached)
7. Note that in upload-err.txt, avrdude runs from a different location than in upload-ok.txt. Yet the avrdude referenced in upload-err.txt can be used successfully from the command line (see cmdline.txt, attached).
8. Remove Arduino megaAVR Boards and the ATtiny upload works again.
[upload-ok.txt](https://github.com/arduino/ArduinoCore-megaavr/files/3478928/upload-ok.txt)
[upload-err.txt](https://github.com/arduino/ArduinoCore-megaavr/files/3478929/upload-err.txt)
[cmdline.txt](https://github.com/arduino/ArduinoCore-megaavr/files/3478930/cmdline.txt)

per1234, 2019-08-07:
It looks like a duplicate of https://github.com/arduino/ArduinoCore-avr/issues/91

>Yet the avrdude referenced in upload-err.txt can be used successfully from the command line (see cmdline.txt, attached).

The cmdline.txt output shows it's using avrdude 6.2:
```
avrdude: Version 6.2
```

JChristensen, 2019-08-07:
> It looks like a duplicate of [arduino/ArduinoCore-avr#91](https://github.com/arduino/ArduinoCore-avr/issues/91)

It's the other way around, yes? For me, in the IDE, 6.3 worked while 6.2 failed. Yet 6.2 seemed OK on the command line.

per1234, 2019-08-07:
>For me, in the IDE, 6.3 worked while 6.2 failed.

No. The avrdude version used in upload-ok.txt is the one bundled with Arduino IDE 1.8.9/Arduino AVR Boards 1.6.23, which is avrdude 6.3.0-arduino14. The avrudude version used in upload-err.txt is avrdude 6.3.0-arduino16.

>Yet 6.2 seemed OK on the command line.

As is expected. The bug is only in avrdude 6.3.0-arduino16.

JChristensen, 2019-08-07:
Not sure where the 6.2 came from. I cannot duplicate that now. Maybe best to ignore the command line scenario. Original issue remains, installing Arduino megaAVR Boards breaks ATTinyCore uploads. Both report 6.3-20171130 in the IDE but with megaAVR installed, avrdude runs from a different location.

facchinm, 2019-08-08:
@JChristensen the TinyCore doesn't specify any dependency tool in its json
```
        {
          "name": "ATTinyCore",
          "architecture": "avr",
          "version": "1.2.5",
          "category": "Contributed",
          "help": {
            "online": ""
          },
          "url": "https://github.com/SpenceKonde/ATTinyCore/archive/1.2.5.tar.gz",
          "archiveFileName": "ATTinyCore-1.2.5.tar.gz",
          "checksum": "SHA-256:db0b7fca838436f04337d4faad47c109e10add1f8b716d689cec4858c8223afc",
          "size": "5569027",
          "boards": [
            {"name": "ATtiny441"},
.....
            {"name": "ATtiny43"}
          ],
          "toolsDependencies": []
        }
      ],
      "tools": []
    },
```
so the IDE picks (what it thinks to be) the best program that matches `avrdude` when requested.
Since megaAVR core ships with `avrdude-6.3.0-arduino16` (which still suffers from arduino/ArduinoCore-avr#91 ) that tool becomes the preferred one.
I'm going to publish an updated megaAVR core shipping with `avrdude-6.3.0-arduino17` (which fixes the issue).
In the meantime, to overcome your problem, you can modify TinyCore's `platform.txt` so [this line ](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/platform.txt#L112) becomes `tools.avrdude.path={runtime.tools.avrdude-6.3.0-arduino14.path}`

JChristensen, 2019-08-08:
@facchinm thank you for the explanation. I don't actually use the megaAVR core, so my fix right now is to just remove it and I'm happy with that. I only had it installed to work on an issue with one of my libraries that someone had reported. It was still installed when I went back to do some work with the ATTinyCore and I was surprised when the upload failed. That's when I found the conflict between the two cores and raised the issue.

MCUdude, 2019-09-16:
It's fixed in the latest 1.8.4 release!

facchinm, 2019-09-17:
Fixed by core 1.8.5

closed 2019-09-17 by facchinm