arduino/ArduinoCore-megaavr #48: [Nano Every] Incorrect RX/TX pin placement?

MCUdude, 2019-08-19:
I'm working on a pinout diagram for the Nano Every, since there doesn't exist one yet.
It turns out that there might be a hardware issue with the TXD1 and RXD1 pins. Why is the TX pin (digital pin 1, TXD1) placed as the first pin, and the RX pin (digital pin 0, RXD1) placed as the second pin?

Is this on purpose or just a hardware bug that's too late to fix?

[image]

per1234, 2019-08-19:
It's on purpose. The Nano Every follows the pinout of the original Nano:
[image]

MCUdude, 2019-08-19:
Well, that's just terrible research from my side... Thanks, @per1234!

closed 2019-08-19 by MCUdude