arduino/ArduinoCore-megaavr #51: Serial interface locks up on Nano Every

technoblogy, 2019-09-10:
On a Nano Every, if you try and transfer more than about 128 characters from the Serial Monitor to the board via the serial interface it locks up, and the only solution is to unplug the USB cable.

The following program demonstrates the problem:

    void setup() {
      Serial.begin(9600);
      while (!Serial);
      Serial.println("Ready");
    }

    int count = 0;

    void loop() {
      if (count % 64 == 0) {
        Serial.println();
        Serial.print(count);
        Serial.write(' ');
      }
      while (!Serial.available());
      Serial.write((int)Serial.read());
      count++;
    }

Open the **Serial Monitor**, copy a line of text of 128 characters, paste it into the input field, and click **Send**.

I'm running Arduino IDE 1.8.9 on a Mac, and the Arduino megaAVR Boards file version 1.8.3.

It works fine on a Uno WiFi Rev 2.

technoblogy, 2019-09-10:
> the serial starts dropping characters

@facchinm this makes the serial unusable for many applications. Is there any way of fixing this? The SAMD11 has up to 16Kbytes of SRAM - surely that's enough for buffers?

facchinm, 2019-09-10:
@technoblogy it has 16KB of flash and 4KB of RAM, the USB buffers can still be enlarged a bit but not that much

facchinm, 2019-09-10:
I repeated your stress test too with  the firmware from #52 and it doesn't lose any character. If you could retest in your environment and report back it would be great.
Thanks you very much for helping.

technoblogy, 2019-09-10:
OK, I'll test it - thanks!

MCUdude, 2019-09-10:
@facchinm now that we got an early alpha version of Optiboot running on ATmega4809, would it be possible/make sense (in the future) to turn the SAMD11 into a dedicated USB to serial adapter instead, and not having to use the UPDI interface for programming? Maybe just change the USB PID so it would be able to identify it?

facchinm, 2019-09-10:
@MCUdude makes sense, but the procedure  would be a bit convoluted (we need to burn optiboot using the UPDI layer BEFORE updating  the D11, a bit error prone :slightly_smiling_face:  )

MCUdude, 2019-09-10:
> EDIT: if you meant on new boards, it TOTALLY makes sense

Yes, doing so for old boards would not make sense, I meant new boards. What's awesome about this is that it would now be possible to use avrdude directly, since the 1200bps touch is difficult to perform outside Arduino IDE. If you went for a 5V compatible USB to serial adapter, like the 8u2/16u2 you can also get rid of all level shifting too.

hiduino, 2019-09-11:
Observations from testing firmware from #52, it seems to only work right after an IDE Upload.  If you unplug and re-plug the USB then it appears the UPDI doesn't reset the 4809.  Also it still appears to lockup after 7 or 8 of the large buffer inputs.

facchinm, 2019-09-11:
@hiduino can you provide a test case for the large buffers lockup? 

If you unplug and replug the usb without externally powering the board the 4809 obviously resets :slightly_smiling_face: 
The behaviour should match classic UNO/nano (reset on open port)

hiduino, 2019-09-11:
@facchinm , Yes, that seems to correct the large buffer lockup issue now.

To clarify, for the board reset, it doesn't seem to work from the IDE Serial Monitor invoke after you unplug/replug the board back in.  The UPDI reset only seem to work after a fresh Upload.  Could it be the state of UPDI after a replug in?

facchinm, 2019-09-11:
@hiduino got it, I've been able to reproduce and I'm taking a look immediately, thanks!

technoblogy, 2019-09-11:
> If you could retest in your environment and report back it would be great.

@facchinm do I just need to upload a new MuxTO to the board? If so, could you give me a download link for it?

hiduino, 2019-09-11:
@facchinm , Okay, looks like the UPDI reset works correctly now after an unplug/re-plug of the USB.
Great work!

hiduino, 2019-09-11:
@technoblogy , you can go to the pull request #52 page and download the MuxTo.bin file.

closed 2019-09-16 by facchinm

technoblogy, 2019-09-18:
@facchinm I've uploaded the latest MuxTO.bin file to my Arduino Nano Every. I can't find any version number to identify it, but it has a modification date of 12 Sept 2019.

Unfortunately I can't detect any improvement - there is still corruption occurring in the serial interface. My demonstration program earlier in this post gives exactly the same sorts of dropped characters.

Should it have improved these?

facchinm, 2019-09-19:
@technoblogy to identify the new version just upload a sketch with verbose on and you should see these lines.
```
M_MCU:
  boot-loader FW version:        1
  firmware version:              1.07
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.07
  hardware version:              1
```
This firmware fixes your test case for me, with no dropped characters

hiduino, 2019-09-19:
@technoblogy , I've also tested the latest MuxTo and did not experience any drop characters.  I've tried copying over 20 times of the large buffer test and no drop characters observed.  I'm testing on two different Windows 7 Pro platforms using Serial Monitor from the IDE.  Have you try testing with different USB cables or on different systems?

technoblogy, 2019-09-19:
I'll try when I get home at lunchtime.

technoblogy, 2019-09-19:
I get this:
````
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
````

facchinm, 2019-09-19:
@technoblogy you did the right thing but it looks like the firmware didn't update properly. Are you sure you are flashing the right .bin? Make sure you remove the one I sent you at the beginning in the zip.

technoblogy, 2019-09-19:
I've tried again. This is what I get in the terminal:
````
Phobos-644:update_muxto david$ ./bossac --port=/dev/tty.usbmodem1411 -U true -i -e -w -v MuxTO.bin -R
Atmel SMART device 0x10030000 found
Device       : ATSAMD11D14AM
Chip ID      : 10030000
Version      : v2.0 May 29 2019 12:13:04
Address      : 4096
Pages        : 192
Page Size    : 64 bytes
Total Size   : 12KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Erase flash
done in 0.631 seconds

Write 12220 bytes to flash (191 pages)
[==============================] 100% (191/191 pages)
done in 7.166 seconds

Verify 12220 bytes of flash
[==============================] 100% (191/191 pages)
Verify successful
done in 0.487 seconds
CPU reset.
Phobos-644:update_muxto david$ 
````
but when I upload a sketch from the Arduino IDE I still get:
````
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
````
Please can you give me a direct link to the correct MuxTO.bin file.

facchinm, 2019-09-19:
The update should read
```
[martino@arch-elitebook Documents]$ ~/.arduino15/packages/MattairTech_Arduino/tools/bossac/1.7.0-mattairtech-3/bossac --port=/dev/ttyACM0 -U true -i -e -w -v MuxTO.bin -R
Atmel SMART device 0x10030000 found
Device       : ATSAMD11D14AM
Chip ID      : 10030000
Version      : v2.0 May 29 2019 12:13:04
Address      : 4096
Pages        : 192
Page Size    : 64 bytes
Total Size   : 12KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Erase flash
done in 0.337 seconds

Write 12288 bytes to flash (192 pages)
[==============================] 100% (192/192 pages)
done in 3.016 seconds

Verify 12288 bytes of flash
[==============================] 100% (192/192 pages)
Verify successful
done in 0.384 seconds
CPU reset.
```
The new firmware is 12288 bytes long; try redownloading using this link https://github.com/arduino/ArduinoCore-megaavr/raw/master/firmwares/MuxTO/MuxTO.bin

technoblogy, 2019-09-19:
OK, thanks, now I get:
````
M_MCU:
  boot-loader FW version:        1
  firmware version:              1.07
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.07
  hardware version:              1
````
However, sorry, I'm still getting dropped characters:

[image]

I'm running this on a Mac with Arduino IDE 1.8.9.

technoblogy, 2019-09-25:
Please can you reopen this issue as it isn't solved.

Also, I think it should stay open even when it is solved, so users who have received the boards with MuxTO prior to 1.0.6 can find out how to upgrade.

jandrassy, 2020-01-19:
I updated the firmware in SAMD11 too. I am on Linux.

Important is to use bossac version 1.7.0-mattairtech-3 installed with MattairTech_Arduino SAMD 1.6.17. The bossac installed with Arduino SAMD and older MattairTech versions report unknown device.

The boards package JSON for MattairTech_Arduino SAMD is hard to find in the middle of the README so here it is: `https://www.mattairtech.com/software/arduino/package_MattairTech_index.json`

the latest firmware bin is here https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO