arduino/ArduinoCore-megaavr #53: No need for the F() macro on the the megaavr architecture

MCUdude, 2019-09-13:
I'll leave this here:
https://github.com/SpenceKonde/megaTinyCore/issues/81

In short terms, it has no RAM usage benefits, but it wastes flash memory. Should probably be defined as `#define F(string_literal) (string_literal)` instead.

facchinm, 2019-09-16:
Totally agree on this; would you like to open a PR so I can give the right attribution?
Thanks

MCUdude, 2019-09-16:
Sure, but that PR has to go into the Arduino API repo (String.h IIRC). I thought this repo should be kept architecture-neutral? Because this is very much an architecture relate thing.

facchinm, 2019-09-16:
Correct; probably the right approach is to `#undef` F macro after including ArduinoAPI.h and redefine it as no-op

closed 2019-09-16 by facchinm

habazut, 2021-01-23:
Did this https://github.com/arduino/ArduinoCore-megaavr/commit/a0f6beb55e6c60fdd7baffb306d99c0844377f43 (which reverted the fix) break this again or do you have another solution for the F() Flash Macro which is present in many codes?
Regards,
Harald.

facchinm, 2021-01-26:
Hi @habazut ,
in fact a0f6beb "broke" this again. We couldn't come up with a clever solution that fits all the situations.
Anyway, in the process we added a bunch of F() relates examples to the CI, just to make sure we are not breaking anyone's sketch again, so if you have any proposal just file a PR and it will be checked against all corner cases.

commit d316aa62 (2019-09-16): Redefine F() as no-op

megaAVR 0-series have a unified address space so forcing a const string in flash just increases flash occupation with no benefit on RAM usage.
Fixes #53