arduino/ArduinoCore-megaavr #63: Serial interface problem with Arduino Nano Every

laurencebarker, 2019-12-18:
I have a problem with the serial interface on an Arduino Nano Every. I've found reference to this problem elsewhere and the solution is to reflash the Muxto. The issue #51 here tells you how to do that for a Mac.

Are there any instructions on how to do this on a windows computer please? I can't get the IDE to compile the muxto.ino file (I assume I need a different board installed). There is a muxto.bin file, but I don't know how to get that downloaded.

I saw that I need to boot the Arduino with two pads shorted; I've done that and it started up differently from normal, with a different com port; but the IDE shows it as "unknown board".

facchinm, 2019-12-19:
Hi @laurencebarker ,
the "unknown board" is expected since the bootloader needs a special core to modify it.
Since you just need to update you can use bossac.exe from [this comment ](https://github.com/arduino/ArduinoCore-megaavr/issues/51#issuecomment-529469380) then follow the latest instructions, using the bin from here (https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO)
Closing the issue, reopen if needed or if it's not clear enough

closed 2019-12-19 by facchinm

MCUdude, 2019-12-19:
I will assume some/many users have early MuxTo firmware in their Nano Everys, and it's not 100% straight forward to flash it if you're new to this kinda stuff. @facchinm why not create a new "Arduino core" just for flashing the MuxTo firmware that can easily be installed? I've done similar with my "Arduino core" for [flashing regular Pro Micros with mEDBG firmware](https://github.com/MCUdude/microUPDI). All it does is to upload a pre-compiled hex and eep file.