arduino/ArduinoCore-megaavr #64: Update toolchain/Atmel packs

MCUdude, 2019-12-22:
Atmel/Microchip has just released a new pack for ATmega devices; 1.4.346. This fixes some naming errors I've stumbled across earlier when working with the CCL peripherals. The latest release also fixes incorrect vector tables for ATmega808/809, and error [I found back in october](https://github.com/MCUdude/MegaCoreX/issues/43) without knowing precisely what caused it.

There's a [whole thread over at AVRfreaks](https://www.avrfreaks.net/forum/atmega809-wont-generate-tca0-cmp-interrupt) about the vector table issue if you want to read more. I would be very grateful if you could apply this patch 🙂

MCUdude, 2020-01-15:
@facchinm is this something you can do? The current version Arduino IDE uses also has some copy/paste errors in the iom4809.h (and others) related to the CCL functionality.

facchinm, 2020-04-07:
I just rebuilt the toolchain with latest packs for both atmega and attiny.
The toolchain is available to be tested as `7.3.0-atmel3.6.1-arduino6` by adding `http://downloads.arduino.cc/packages/package_megaavr_index.json` to Additional json.
It will prompt to install the fake 1.9.9 version of megaAVR core, so the toolchain will be downloaded as a dependency.
@MCUdude let me know if it is packaged correctly :wink:

MCUdude, 2020-04-07:
I just installed 1.9.9, and it now compiles using 
```
/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino6/bin/avr-gcc
```

Everything seems good. Thank you!

PS. is there a lot of work to include the new AVR DA pack as well? 🙂 That new hardware looks even more hardware, and I'd love to play around with it in Arduino IDE!

facchinm, 2020-04-07:
Not a lot of work but the procedure is a bit convoluted due to the atpacks license :slightly_smiling_face: 
If you want to try, copying the build script from https://github.com/arduino/toolchain-avr/blob/staging/atpack.tiny.build.bash and modifying it should ease the procedure

MCUdude, 2020-04-07:
I looked through the atpack, but couldn't find any references to a license. Would you like to point me towards it?

The AVR128DA48 looks like a killer chip, and would be perfect for a "Pro" Nano Every!

facchinm, 2020-04-07:
Eheh the license is the one you must accept when downloading the atpack (so there's no easy+legal way to bypass it and the files must be copied manually in the toolchain build folder).
AVR128DA48 looks cool btw :wink: Still no native USB unfortunately.

xoriath, 2020-07-08:
> Eheh the license is the one you must accept when downloading the atpack (so there's no easy+legal way to bypass it and the files must be copied manually in the toolchain build folder).
> AVR128DA48 looks cool btw  Still no native USB unfortunately.

Sorry for barging in... But, how could we make it easier to get the atpack..? The intention of the pack server was never to be something  user facing, just a cmsis compliant repository (I.e the web UI is just something similar to the Keil site).

Hit me up somewhere if you want to talk :smile:

mcuee, 2023-06-04:
This is a very old issue but now there are Arduino 3rd party core to support UPDI AVR chips.

https://github.com/MCUdude/MegaCoreX (you can use Nano Every using this one)
https://github.com/SpenceKonde/DxCore (you can use AVR DA/DB/DD parts)
https://github.com/SpenceKonde/megaTinyCore (you can use UPDI Tiny AVR0/1/2 chips)