arduino/ArduinoCore-megaavr #66: TWI in slave mode using port PF2 and PF3

rlivio, 2020-01-28:
I need to use the ATmega4809 with I2C configured in slave mode using pins PF2 and PF3.
The only way to obtain that is to add: 

`PORTMUX.TWISPIROUTEA |= TWI_MUX;

TWI0.DUALCTRL = TWI_ENABLE_bm;`

after the `Wire.begin(4);`

Probably the instruction: `PORTMUX.TWISPIROUTEA |= TWI_MUX` is missing inside the function TWI_SlaveInit() in file twi.c

Those who have more experience than me can check.

Livio

tomalmy, 2020-10-20:
I've used the TWI on a Nano Every board both master and slave simultaneously (it takes some mods to twi.c to re-enable  that functionality). But in answer to your question, because PA2/PA3 and PF2/PF3 are bonded together on the board it is not necessary to set PORTMUX or DUALCTRL, nor will it make any practical difference. The TWI master and slave interfaces will always share the same SDA/SCL pins on the board.

curiosity4809, 2021-09-13:
Hey tomalmy,

can you tell me which functionalys in twi.c youve made to use master and slave in same time?

tomalmy, 2021-09-13:
The issue is that twi_mode is used for both master and slave operation. And executing either TWI_MasterInit or TWI_SlaveInit precludes executing the other. 

I added a new variable static volatile TWI_MODE_t twi_smode; Then all the slave functions are changed to use twi_smode instead of twi_mode. Also in TWI_SlaveInit change the last line to:
`	TWI0.MCTRLA |= TWI_ENABLE_bm; // PATCH change to |= so as not to disable Master operation
`