arduino/ArduinoCore-megaavr #83: Wire library has no timeout functionality (meaning it is blocking).

second-pair, 2020-09-09:
In trying to solve some I2C issues I've been having, I came across [this](https://github.com/arduino/ArduinoCore-avr/pull/107) thread (and a great many others), which indicate the Wire libraries don't have any timeout capabilities, particularly making `Wire .endTransmission ()` blocking.  This has [since been solved and merged](https://github.com/arduino/ArduinoCore-avr/pull/107#issuecomment-642671632) into _ArduinoCore-avr_ and has appeared in my libraries locally.  To the best of my knowledge (and my own experience), the _ArduinoCore-megaavr_ Wire library is also blocking.  I haven't been able to find this fix in the _ArduinoCore-megaavr_ version of Wire, and I haven't found any other issues relating to this, so I'm raising one now.

I'm not aware of the development relationship between the two cores, so if this already on its way, that's great.  If not (and it's appropriate to do so), I'd like to see the functionality added to _ArduinoCore-avr_ added here too.  Particularly, I'd like to see functionality added that's similar to that provided by the new functions `setWireTimeout()`, `getWireTimeoutFlag()`, `clearWireTimeoutFlag()`.

freddyrios, 2020-10-05:
Also interested to hear if this also affects the megaavr. I spotted a while loop in the source, so it seems likely.

Do note however that this is a (good) mitigation to try to reduce the impact of underlying issues. So we are supposed to look into does regardless of this change. See https://github.com/arduino/Arduino/issues/10803#issuecomment-702905614

softhack007, 2021-11-27:
Hi, from looking into the wire library code, it seems to me that the logic is very different from arduinocore-avr. The only place where a "while" loop occurs is in function TWI_MasterWriteRead():

https://github.com/arduino/ArduinoCore-megaavr/blob/c31614b921f4c44eb2b1a2d19b7b2f172fcb6b86/libraries/Wire/src/utility/twi.c#L327-L330

In contrast to the avr core, the main work is performed by hardware and an ISR. 

So the first question is- is it possible that the ISR never updates the status? I saw that there are error codes for "bus error" and "arbitration lost".

valerioformato, 2023-04-05:
+1 I find the I2C bus on my Uno wifi rev2 randomly freezing with 2 devices on it, and this feature would probably help a lot