arduino/ArduinoCore-megaavr #88: MLX90614 I2C communication on Arduino Nano Every

omersa, 2020-12-02:
I have bought 6pack of Arduino Nano Every and built a splecial shield plate, so I can connect thermal sensor MLX90614, which communicate over I2C (A4,A5). The sensor was well tested on few Nano devices and on all it's OK. But with Arduino Nano Every it does not work. The device can be found on port I2C (search address), but no communication is going through. No data can be read or write. Tried many libraries and the problem persists with all.

As this device is has different processor then classic Nano, it seems that libraries for megaAVR boards are not OK. Also I am not alone with this problem. I think it persist on all Wire communication with Every.

omersa, 2020-12-04:
I have done some deep search into sources and I have found out that Wire and Twi libraries for megaavr (Arduino Nano Every) are completelly different from avr libraries that work on Nano. They are much older with no timeouts and many other changes. It seems that until somebody renew the libraries for megaavr boards, ther will be no solution. Have tried to take libraries from avr, but does not work, as it seems there is much difference in all other libraries.

alranel, 2020-12-04:
@omersa, we are definitely going to check this - give us some time and we'll get back to you :)

alranel, 2020-12-06:
Some possibly related issues, discussing the "repeated start" behavior required by this sensor:

https://os.mbed.com/forum/mbed/topic/586/
https://electronics.stackexchange.com/questions/275496/problem-i2c-nack-in-3rd-byte-mlx90614-repeated-start
https://forum.arduino.cc/index.php?topic=137607.0

alranel, 2020-12-06:
I got a GY-906 breakout board like yours and I was able to reproduce the issue. I did these tests:

* ✅ Arduino Uno: works fine
* ❌ Arduino Nano Every (with official megaAVR board package): does not work
* ✅ Arduino Nano Every (with [MegaCoreX](https://github.com/MCUdude/MegaCoreX) board package): works fine

omersa, 2020-12-06:
You right. I manage to make it work with MegaCoreX. Thank you.

omersa, 2020-12-07:
I changed the twi.c and - YES, it works fine now. Thank you for all your help.

alranel, 2020-12-07:
Thank you for reporting the issue @omersa!

closed 2020-12-09 by facchinm

Nolan-Walker, 2021-07-25:
Hi there @alranel , I am having the same issue as above but the change is not working for me. I have the MLX90614-DCI version of the sensor without the PCB ("naked version"); I have also soldered a 100nF capacitor between VSS and VDD... When using the regular Arduino Nano the sensor works fine, but when I use the Arduino Nano Every, the sensor does not work and returns "nan". I have the updated version of the arduinocore - megaavr board package and I also have tried using MegaCoreX board package and neither of these are fixing the issue for me.

Any help would be greatly appreciated! Thank you for your time!

PR #90 (2020-12-09): Bugfix: wrong default I2C baud rate was set for Arduino Nano Every
This PR fixes #88.

Whenever the default baud rate was used, or a custom frequency being less than 800kHz or greater than 1,200kHz was set, the computed baud rate was incorrect due to an integer overflow.
`F_CPU_CORRECTED` is a `uint32_t` set to 16000000 for Arduino Nano Every. It gets multiplied by `t_rise` which is set to 1000, thus it overflows the numeric range for a `uint32_t`.
When using the default baud rate, `TWI0.MBAUD` was set to 73 instead of 67 which appears to be the right value.

(How can we add a unit test for th [... cut]