arduino/ArduinoCore-megaavr #89: Wifi Rev2 Upload Issues (verification error;content mismatch)

nametable, 2020-12-03:
I am having an issue with uploading sketches/applications that are bigger than around 512 bytes to my Arduino Uno Wifi Rev2. I think that the issue is likely related to incorrect mEDBG firmware on the 32u4. I was reprogramming the 32u4 to use it as a HID and then wasn't sure what to reprogram it as to make it work again to program the 4809. 

I have tried flashing the 32u4 with `medbg_updi.hex` from Atmel Studio (tools/medbg_fw.zip) and also firmware provided with Xplained Yourself (https://hackaday.io/project/162372/files), whose UPDI .hex file seems the same as what is provided with Atmel Studio. Currently my 32u4s EEPROM has been programmed with the .eep from Xplained Yourself. The original EEPROM has been erased. I have noted that with only 0s in the EEPROM, no programming of the 4809 works at all.

Below I have included an AVRDUDE log including the error (the intended program never seems to run as no Serial communication is detected after programming):
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/logan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/logan/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: usbdev_open(): Found mEDBG CMSIS-DAP, serno: XYS by Vidmo UPDI/04
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: max packet size expected 512, but found 64 due to EP 0x81's wMaxPacketSize
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.13 (rel. 43)
         Serial number   : XYS by Vidmo UPDI/04
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: erasing chip
avrdude: reading input file "/tmp/arduino_build_614042/Scan.ino.hex"
avrdude: writing flash (24244 bytes):

Writing | ################################################## | 100% 7.46s

avrdude: 24244 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_614042/Scan.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_614042/Scan.ino.hex:
avrdude: input file /tmp/arduino_build_614042/Scan.ino.hex contains 24244 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 6.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0200
         0x08 != 0x0c
avrdude: verification error; content mismatch
avrdude: jtag3_edbg_signoff(): unexpected response 0x03, 0x80

avrdude done.  Thank you.
```
Is there a way to get the original firmware and a typical EEPROM back? I'm not 100% sure that the firmware is the issue but that is really the only thing I changed (via dfu-programmer). I find it odd that I can program small programs but not ones that are larger (mismatch at 0x200)... I almost wonder if somehow the 32u4 is now programming the wrong part of flash in the 4809 or if some fuse may be out of wack...

Any help reseting/fixing my Uno Wifi Rev2 would be appreciated.

manilal2, 2021-01-29:
please have a look at https://github.com/arduino/ArduinoCore-megaavr/issues/33, I am working with custom made PCB and I am not able to put atmega4809 in program mode , once I was able to erase it and program it , after that it does not program , any help fixing it would be appreciated

manilal2, 2021-01-29:
here is picture of error which I am getting 
[image]

manilal2, 2021-03-10:
I finally got bootloader burnt on my custom made PCB, when I try to program using arduino IDE I am getting content mismatch at address 0x200 ( which is 512 in decimal ) , verification is failing I am using following bootloader (Optiboot_mega0_UART1_DEF_115200_A7.hex) with internal 20 Mhz clock, UART1 on default pins , 48 pin, RESET pin, 

why I am getting content mismatch after programming during verification stage  ?

nametable, 2021-03-10:
This error is the same or similar to the issue I've had except I was not using OptiBoot, just medbg from Atmel Studio. I believe the mismatch is due to corruption when flashing. In my testing, after flashing greater than the first 16KiB, corruption occurs. I've described this phenomenon here: https://forum.arduino.cc/index.php?topic=717504.0 .

I am still wanting to know a solution as well...

MCUdude, 2021-03-10:
Do you guys get the same error if you program your devices with a different programmer, for instance, JTAG2UPDI or an mEDBG variant such as [microUPDI](https://github.com/MCUdude/microUPDI)?

manilal2, 2021-03-10:
I am using Atmel-ICE ( UPDI) to burn bootloader and then I am using arduino megaCoreX to load sketches and I am getting content mismatch error , I am not getting any error when uploading bootloader using Atmel-ICE ( updi)

manilal2, 2021-03-12:
I just tried microUPDI ( mEDBG) , here is error when trying to read signature 
[image]

I am also unable to erase chip, here is error which I am getting 
[image]