arduino/ArduinoCore-megaavr #100: Compiling fails when using precompiled library

landret, 2021-04-19:
Hi,
Compiling fails at linking when using a precompiled library .a for atmega4809 targets. The following message appears in the verbose:

"Using precompiled library in C:\... (archive library path)
The plaform does not support 'compiler.libraries.ldflags' for precompiled libraries."

Looking at the verbose, the library path (-Lfolder -lname) is missing from the linking command line.

Is that a normal behavior and could this be solved? 
Thanks

facchinm, 2021-04-19:
Hi @landret ,
precompiled libraries are not supported in this core yet. A patch like https://github.com/arduino/ArduinoCore-avr/commit/bca2493a5c68ba5c0a2c6963b462d917794bfb2d should be enough but needs extensive testing. Would you mind creating a PR? Thanks!

landret, 2021-04-19:
Hi @facchinm,
thanks for you quick answer. I successfully tested the patch with both Nano Every and Uno Wifi rev2. I will submit a PR for this patch.