arduino/ArduinoCore-megaavr #102: Arduino Nano Every no Serial Communication after Start

NelsonIg, 2021-04-30:
Hey everyone,

I try to communicate with the Arduino Nano Every through the serial usb port. This works fine when using the Arduino Serial Monitor. But if I use a different monitor/device, the arduino nano every will not respond. Only after first connecting the arduino serial port with the Arduino Nano Every, it works fine.

So how can i communicate to the Arduino Nano Every right after start without connecting it first to the Serial Monitor of Arduino.

sample code:
```
void setup() {
  Serial.begin(9600);

}
int incomingByte = 0; // for incoming serial data
void loop() {
  // send data only when you receive data:
  if (Serial.available() > 0) {
    // read the incoming byte:
    incomingByte = Serial.read();
    
    Serial.write(incomingByte);
    Serial.write(incomingByte);
  }
}
```

Thanks in advance!

thomome, 2021-05-17:
I probably encountered the same bug. 
Since the Arduino SerialMonitor has dtr enabled by default, the Arduino Nano Every auto-resets and works properly. But with dtr disabled `nona4809.serial.disableDTR=true` in boards.local.txt I am not able to get the serial port communicating on any other baudrate than 115200 (default of the firmware?)

@NelsonIg If you have control of dtr in your monitor/device, you could try to enable it or switch to 115200 baudrate as a workaround. 

Since I don't have control over baudrate and dtr of the client, I would be very grateful for a fix.

NelsonIg, 2021-05-20:
Thanks you.  This helped! Don't know how this wasn't answered before

closed 2021-06-17 by NelsonIg