arduino/ArduinoCore-megaavr #103: Compiled MuxTo binary to large

thomome, 2021-05-25:
Hello everyone,

I was trying to compile and upload the MuxTO firmware for the Arduino Nano Every. I followed the guide https://www.arduino.cc/en/Guide/NANOEvery and was able to compile the sketch with arduino-cli.
```bash
arduino-cli compile -e -b MattairTech_Arduino:samd:muxto:float=default,config=enabled,clock=internal_usb,timer=timer_732Hz,bootloader=4kb,serial=two_uart,usb=cdc
```

After exporting the binary, I noticed that it was 280 bytes larger than the precompiled firmware binary from the repo. When I anyway tried to upload it with bossac it failed with the following error. 

[image]

Does anyone have any idea what I am doing wrong?
Any help would be appreciated.

facchinm, 2021-05-27:
Hi @thomome ,
the project grew a bit since the instructions were last written (I should have checked it, my fault).
The easiest way to make it fit again in 12KB of flash is to use a newer compiler. LTO should be a viable solution but some patches have not been ported to that core, so it's unlikely to compile out of the box.
For the records, arch linux default arm-none-eabi-gcc compiler works just fine :slightly_smiling_face:

per1234, 2021-05-27:
>I should have checked it

I can set up CI for it if you like. The workflow I wrote for the sketch compilation "smoke test" of the Motor Carrier firmware can be a good reference:
https://github.com/arduino-libraries/ArduinoMotorCarrier/blob/master/.github/workflows/compile-firmware.yml

facchinm, 2021-05-27:
@per1234 it would be great! The only step to add is to check that the final .bin is smaller than 12288 bytes (0x3000), since the size report is reporting a slightly smaller size.

per1234, 2021-05-30:
Hi @facchinm. I'm afraid I don't understand the requirement. Shouldn't the compilation process fail already if the compiled sketch size exceeds the `upload.maximum_size` property value for the board?:

https://github.com/arduino/ArduinoCore-samd/blob/5c850ac24d9d2b5ea0b322d3fb44dbfd0b63606f/boards.txt#L1617
```
muxto.menu.bootloader.4kb.upload.maximum_size=12288
```
I can certainly add another step to the workflow to run `arm-none-eabi-size` on the compilation output, and a regex on the output of the command, but it doesn't seem like it should be necessary since Arduino CLI does that already.

facchinm, 2021-06-07:
Basically `arm-none-eabi-size` doesn't take into account the vectors section, resulting in a smaller number than the actual size that we'll then try to upload (and fail).

ganghuang, 2021-12-29:
@facchinm 

As this issue shows that the latest version can not compile successfully for the size of arduino nano every.
Would you please help to point out which commit in this repo can be compiled successfully for nano every with 4k bootloader?

And also who compiled the "factory" version of the firmware in the purchased nano every board? Which commit does it associated with?

Thanks,

ganghuang, 2021-12-30:
I modified the MuxTO.ino as attached. 
Now the project can compile to 12284, the size of nano every need <12288; 
So I can upload the bin to the SAMD11D14A on the nano every.

But after upload I can not load the regular sketch and I got the error 
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
[MuxTO.ino.txt](https://github.com/arduino/ArduinoCore-megaavr/files/7791302/MuxTO.ino.txt)
And also the diff to the MuxTO.ino in repo as of 4137a59 ( which is same file as e458d1d)
[MuxTO.ino.diff.txt](https://github.com/arduino/ArduinoCore-megaavr/files/7791311/MuxTO.ino.diff.txt)

Basically, I did the following change:
1. move global variables to the loop function except the serial_mode, baudrate and lock q;
2. we are not using the LED_BUILDIN here, 
3. change the two updi_mode if to if/else structure
4. For several functions which has been called multiple times, combine them to one single call and give the result to a variable
5. Use static const for the mask and bit set for different serial new value
6. Combine the serial_mode & mask and |= new value together

For me it doesn't seems I have changed the behavior of the code, but I am not sure, especially it is not working yet.

ganghuang, 2021-12-30:
>

ganghuang, 2021-12-30:
Item 1 in my modification doesn't work as the code need data between iterations.

ganghuang, 2021-12-30:
Since my last modification is wrong, I am thinking about what can we do to reduce the size. 
I think there are two possible directions:
1. Reduce the size of crc table by change it to calculation
2. Reduce the unused hardware support

mcuee, 2022-08-17:
Unfortunately the github action never runs successfully due to this issue.
https://github.com/arduino/ArduinoCore-megaavr/pull/105

facchinm, 2022-08-24:
I got it fixed by removing some unused features from the core and, most important, updating the compiler version (that was stuck to 4.9.x)
This is the PR https://github.com/arduino/ArduinoCore-megaavr/pull/122 ; it should generate the correct bin (under Artifacts)

closed 2022-08-24 by facchinm

chalupec, 2023-03-17:
Hello, i have loaded successfully last version, i have replaced CRC table with calculation. I have encountered errors on UART reception. 115200baud speed. Transmittion from atsam->atmega works great(perfect timing and levels), also atmega->atsam works (checked by external receiver, perfect timing and levels), but atsam passes invalid data to serial terminal in PC. Basically i am sending "POS 1|1|1|1\r\n" and i get "POScddd>d" in terminal or "Pdcdddkd".. Does anybody encountered something similar?

TtomGit, 2023-07-25:
@facchinm I seem to be working out of my depth here, But I am having this same problem, though it should be solved. I have put together a custom nano every and would like to upload the factory program to the ATSAMD11D14A, but am stuck here. I have followed the instructions from in0 on this page: https://forum.arduino.cc/t/reset-nano-every-via-1200-baud-touch/939949/8 but have stumbled over a few issues.

Firstly, the location online for the firmware and instructions:  https://www.arduino.cc/en/Guide/NANOEvery#firmware-for-samd11d14a seems to be missing. But ok, I am trying to work around.

Secondly, the versions of "MattairTech SAM D|L|C core for Arduino" that are currently available do not allow for 4kb bootloader.  I was able to find "Fab SAM Core for Arduino" seems to be able to do this, however I have not been able to test this yet.

Thirdly, when trying to complile MuxTO, the flash is overflowed by 424 bytes. 

I am using the Arduino IDE 2.1.1. 

Really what I need are explicit instructions for how to program the SAMD11D14A so that it can be used to program the ATMEGA4809. 

If there is any way that you can help or if you have any comments about the issues I am having it would be greatly appreciated.

facchinm, 2023-07-26:
@TtomGit the [Compile MuxTo github action](https://github.com/arduino/ArduinoCore-megaavr/actions/workflows/compile-muxto.yml) already creates a zip file containing a compiled firmware as artifact :slightly_smiling_face: 
You can fork this repo, apply the patches to MuxTO sources, push on your fork and the CI will produce the correct firmware. Let me know if it helps!

TtomGit, 2023-07-26:
@facchinm Thank you, for your quick response. I created a new branch and used the Compile to MuxTo action which produced a binary file that I uploaded to my custom Arduino Nano Every and also a factory Nano Every to check if it was a software or a hardware issue . However when I connected via USB to both, nothing was detected over any of my COM ports. I tested the factory Nano Ever before I flashed the binary file and it was detected, so it is not a issue with my computer or the hardware, but rather the firmware. You had mentioned applying patches to MuxTO sources, but I am still rather confused what you mean by that. What do you mean by patches? Are there any in the repository or elsewhere? Because I am only trying to upload the factory firmware. Is that not what is created with the current source files? Thank you, for your help.