arduino/ArduinoCore-megaavr #108: Slightly misleading "compatibility spec"

RoiEXLab, 2021-09-10:
Hi everyone,
not sure if this is the right place to post this, but I thought I'd try anyways.
Basically I was browsing through available libraries for my nano every board, and noticed the "HID for megaavr" library, which explains how the library can be used for USB keyboard, mice, gamepads etc.
To my disappointement I got the "no native USB" error message from the macro, guess I'll have to buy another board for this then (even though technically the board has another USB chip on it that could in theory be used as well, but that's a different story).
I assume this compatibility information is derived from the architecture value specified here: https://github.com/arduino/ArduinoCore-megaavr/blob/8e89372cc6ff183c0d97c4d5e4feceb09de03088/libraries/HID/library.properties#L9
But I think this particular case shows that it needs to be a little bit more specific than that, especially for beginners this could be pretty misleading tricking them into thinking their board can do stuff that it clearly can't just to be greeted with a macro error.

TL;DR the HID library is displayed as "supported by nano every, Uno WiFi Rev2" in the arduino web IDE even though it clearly isn't supported, and I think this is misleading and should be fixed.

facchinm, 2021-09-13:
@arduino/ft_content

per1234, 2021-09-13:
@facchinm what is the purpose of the HID library in this repository?

facchinm, 2021-09-13:
Oh wow, none, it's just a leftover from the `avr` core legacy. We can safely remove it

closed 2021-09-22 by facchinm