arduino/ArduinoCore-megaavr #110: [Technical Question] USB passthrough possible with custom firmware?

RoiEXLab, 2021-09-13:
So I'm wondering about the feasibility of this idea for a while now, and I thought I could ask here to get some facts straight.

If I understood this correctly, the Arduino Nano Every has 2 "arduino-compatible" chips on it, right?
The ATMega4809 (main chip) and the ATSAMD11D14A which is used to reprogram the former, whenever new software is flashed onto the board.
To ensure communication can happen through the serial interface to an attached computer, both chips continue to communicate over a serial connection with each other if I understand this part of the code correctly:

https://github.com/arduino/ArduinoCore-megaavr/blob/8e89372cc6ff183c0d97c4d5e4feceb09de03088/firmwares/MuxTO/MuxTO.ino#L72-L90

Only the ATSAMD11D14A has native USB support, and if I understand the datasheet correctly USB communication works by writing to the correct registers and letting the hardware handle the rest. (Maybe oversimplified?)

So is there anything that would speak against extending the MuxTO firmware for the ATSAMD11D14A to allow the serial interface to use some sort of protocol that allows the ATMega4809 to write data into those registers?
It would obviously introduce some overhead, but on the other hand it would allow people to use their Nano Every as a proper USB device, with some limitations.

Is there any fundamental flaw I'm missing here? Or is it just something that nobody bothered doing so far? Also I'm not sure how exactly arduinos communicate with a PC over USB by default, so maybe there is no way to still flash new software using the IDE or CLI after changing the firmware?

I'd be happy if you could grant me any insights on this topic, for now I'm just really curious.

facchinm, 2021-09-22:
Hi @RoiEXLab ,
first of all, thanks for your interest in this topic!
Basically, on the D11 we run a proper arduino sketch (compiled with a patched version of this core https://github.com/mattairtech/ArduinoCore-samd) so it's teoretically possibile to extend it by adding more functionalities.

There's a catch tough; the flash in the chip is very limited so adding "stuff" is likely going to blow the memory.

One solution would be getting rid of the bootloader (which takes 4KB and allows reprogramming the D11 firmware without a programmer). This would give some space to create the infrastructure you have in mind.

If you want to embark in this adventure I can give some more hints on the things that need to be exported and how to handle them :wink:  
Maybe you could get some support by writing to the forum too (@NicoHood did a similar great job on the UNO 16u2 for example)

NicoHood, 2021-09-22:
[Hoodloader2](https://github.com/NicoHood/HoodLoader2) is written for AVR. It looks like that is not trivial, but I am really not into that topic.

RoiEXLab, 2021-09-24:
First of all thanks for your answers, they are very appreciated.

A couple of questions still remain though:

1. Where does the `USBCON` macro used everywhere come from? If I understand it correctly it indicates if the microprocessor has some sort of USB registers. Which makes me wonder though: How is the `Serial` object/interface for the ATMega4809 implemented? The CDC Header defines this object here https://github.com/arduino/ArduinoCore-megaavr/blob/f21b3ae145765bb2beeddd693fe8ab1d528b8bf1/cores/arduino/CDC.h#L89 but the cpp file has the implementation behind a USBCON macro check? https://github.com/arduino/ArduinoCore-megaavr/blob/f21b3ae145765bb2beeddd693fe8ab1d528b8bf1/cores/arduino/CDC.cpp#L27 So wouldn't this mean that boards without native USB can't use serial, or is this class implemented elsewhere and this is just another unused code relic, just like in #108 ?
2. What's the difference between the fork of https://github.com/mattairtech/ArduinoCore-samd and the original https://github.com/arduino/ArduinoCore-samd ? Or is the fork the "patched version" you were talking about?
3. How would I compile MuxTo and the megaavr-core code and upload it to the arduino? Looks like the arduino IDE seems to be the easiest option, but it doesn't provide a "muxto" board as a compile target, or is the only way using arduino-cli or even manually using avr-gcc? Any guidance for your preferred setup?
4. Also in case I actually get something working in the end and I don't abort this project out of frustration after a week, is this something that could make it into the official code to be used by newer boards or will it stay unofficial forever?

Thanks in advance for your patience!

RoiEXLab, 2021-09-24:
Ah nevermind about 2. and 3. found the article: https://www.arduino.cc/en/Guide/NANOEvery