arduino/ArduinoCore-megaavr #116: Datasheet for Nano Every/Wifi Rev 2 entries for I2C and A4/A5, A6/A7 query?

diyelectromusic, 2021-12-14:
The datasheet/pinout for the Nano Every shows A4/A5 as PA2/PA3, but from my reading of the ATMega4809 datasheet, these are not the I2C port that will also support the ADC.  The code in pins_arduino.h suggests these are actually PF2/PF3 which does support both I2C and AIN12/13.

Similarly, for the Wifi Rev 2, A4/A5 are listed as PD4/PD5 (18/19) but the SDA/SCL additional pins are also listed as (18/19) but on PA2/PA3.

Is this correct?

facchinm, 2021-12-15:
I just checked the datasheets and in fact you are right.
The dual I2C/analog functionality is obtained by shorting the pins internally (can be seen in the actual schematics).
For example, in the Nano Every, A4/SDA is connected to both PA2 and PF2, while  A5/SCL is internally routed to PA3/PF3

@per1234 can you assign it to the right person to fix the pinout files? Thx :slightly_smiling_face:

per1234, 2021-12-15:
I believe @FunzioneFabio is our resident pinout diagram wizard. Are you able to help us out with this @FunzioneFabio?

FunzioneFabio, 2021-12-21:
Thank you @diyelectromusic for spotting the error, the pinouts now are fixed and online!