arduino/ArduinoCore-megaavr #118: Nano every reset

ganghuang, 2021-12-27:
I can not get the nano every reset by sending the 1200 baud signal and I am thinking this might because of the atsamd11d14a is not doing that.

I noticed the latest commit on the MuxTO is to "Reset 4809 on baudrate change / port open". 
I don't really like the reset on port open, but I do like the reset 4809 on the 1200 baud. I am thinking if i can compile the project and load the firmware, then I might be able to modify it to only reset at 1200 baud and not reset at port open.

So I try to compile the firmware, and found out that is not easy:
https://forum.arduino.cc/t/compiling-and-loading-the-muxto-code/627145 
and also I got the file size 
/home/myname/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: MuxTO.ino.elf section `.text' will not fit in region `FLASH'
/home/myname/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 112 bytes
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic D11D14AS.
[image]

By change the Bootloader size to compile with "no bootloader", the GUI can compile. Then I try to upload it

I connected USB with the SWDIO and GND pins as instructed in the getting started (https://www.arduino.cc/en/Guide/NANOEvery#firmware-for-samd11d14a : To reprogram the D11, simply power it while shorting SWDIO to GND pins.)

Then I  have the 
Bus 001 Device 048: ID 2 341:5058 Arduino SA MuxTO BL
and also have 
/dev/ttyACM0

Selecting the Mattair generic D11D14AS
[image]

I can not upload the 
[image]

I am using a debian 11 and arduino 1.8.16. using the latest clone from the git 4137a5 of this repo.

per1234, 2021-12-29:
There has been some progress on the subject over on the Arduino Forum:

https://forum.arduino.cc/t/reset-nano-every-via-1200-baud-touch/939949

ganghuang, 2021-12-29:
> There has been some progress on the subject over on the Arduino Forum:
> 
> https://forum.arduino.cc/t/reset-nano-every-via-1200-baud-touch/939949

Still struggling on getting the basic work on that side too.

mcuee, 2022-08-17:
It seems to work for me.

Reference:
* https://github.com/avrdudes/avrdude/issues/980

You can also try to use avrdude directly and the 1200bps trick mentioned in this thread, using shell script or pyserial or things like that.
* https://github.com/avrdudes/avrdude/issues/788#issuecomment-1142055590