arduino/ArduinoCore-megaavr #123: Deployed platform.txt in v1.8.7 is not the good one

landret, 2022-11-21:
Hi,
when downloading megaavr package v1.8.7 from the Arduino IDE, the platform.txt included is incorrect. It is not the same as on Github. It is missing #101 PR commit that was merged on 26 september 2021.

per1234, 2022-11-21:
Hi @landret. I think the way you have formulated this report is likely to cause confusion.

Your report implies it is unexpected that the `platform.txt` or the 1.8.7 release does not contain the change from https://github.com/arduino/ArduinoCore-megaavr/pull/101. https://github.com/arduino/ArduinoCore-megaavr/pull/101 was created 4 months after the time of [the 1.8.7 release](https://github.com/arduino/ArduinoCore-megaavr/releases/tag/1.8.7), so it is expected that the 1.8.7 release would not contain that change:

https://github.com/arduino/ArduinoCore-megaavr/compare/1.8.7...master

So it seems that what you actually intended was to request a new release be created.

landret, 2022-11-21:
Hi @per1234 
I thought v1.8.7 was released after change from #101. I misunderstood, sorry about that.

Indeed, it would be interesting to have a new release created as the last one happened nearly 2 years ago.

facchinm, 2022-12-21:
Core 1.8.8 contains https://github.com/arduino/ArduinoCore-megaavr/pull/101 , closing as fixed

closed 2022-12-21 by facchinm