arduino/ArduinoCore-megaavr #124: Nano Every on linux: "avrdude: jtagmkII_getsync(): sign-on command: status -1" when uploading sketches

chconnor, 2022-11-27:
Hello -- with IDE 2.0.2 or 1.8.19 running on Kubuntu Linux 22.04 I am unable to upload sketches to the Nano Every.

It gives many errors like this:

`avrdude: jtagmkII_getsync(): sign-on command: status -1`

...eventually followed by:
```avrdude: jtagmkII_getsync(): timeout/error communicating with programmer (status -1)
Failed uploading: uploading error: exit status 1
```

The same sketch/board/USB cable/USB port combination works fine in Windows 10 on IDE 2.0.2.

**I have found, in Linux, that if I change the Serial monitor to 1200 baud (from the default of 9600), and then upload the sketch, it works.** After that I can upload the sketch at any baud rate, including 9600 baud, until the board is power cycled.

I was told this was the place to report this issue. Let me know if I can provide any other information.

Verbose console output, in case it's useful:

```
/home/savel/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino-sketch-014E27C2218B86F33F99FEE7D689DAE1/Telemetrix4Arduino.ino.elf
Sketch uses 21797 bytes (44%) of program storage space. Maximum is 49152 bytes.
Global variables use 1630 bytes (26%) of dynamic memory, leaving 4514 bytes for local variables. Maximum is 6144 bytes.
Performing 1200-bps touch reset on serial port /dev/ttyACM0
"/home/savel/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/home/savel/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega4809 -cjtag2updi -P/dev/ttyACM0  -b115200 -e -D "-Uflash:w:/tmp/arduino-sketch-014E27C2218B86F33F99FEE7D689DAE1/Telemetrix4Arduino.ino.hex:i" "-Ufuse2:w:0x01:m" "-Ufuse5:w:0xC9:m" "-Ufuse8:w:0x00:m" {upload.extra_files}

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/savel/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/savel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
...
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): timeout/error communicating with programmer (status -1)

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1
```

facchinm, 2022-11-28:
Hi @chconnor ,
we found this problem some time ago and it looks like a regression in the linux kernel; basically the "set 1200bps" message and the "open port" were well separated (some milliseconds) while in recent kernels (but not on all distributions) the second is being sent immediately after the first, creating some confusion in the D11 firmware.
Setting the port at 1200bps separately gives the fw some time to properly set its internal state.
We definitely need to allocate some time to fix the issue mainline (or simply add a quirk for the Nano every) beacuse updating the D11 firmware is not viable for most of the users.
@agdl

tehkillerbee, 2022-11-29:
@chconnor We have the same issue. We made a small python script to be executed before attempting to upgrade firmware:

```
# total arguments
n = len(sys.argv)
if n >= 2:
    port = sys.argv[1]
else:
    port = '/dev/ttyACM0'
    #port = '/dev/ttyMCU'

print("Port: {}".format(port))

#re-enable hupcl temporarily (necessary for arduino reset using serial port)
os.system('sudo /bin/stty -F {} hupcl'.format(port))

try:
    #perform Arduino Nano Every reset "handshake"
    ser = serial.Serial()
    ser.baudrate = 1200
    ser.port = port
    ser.open()
    ser.close()

except serial.SerialException:
    print("Error: serial.SerialException")
    exit()
```

gratefulfrog, 2024-05-07:
This is an ongoing issue and quite painful.

I learned that setting the serial monitor to 1200 baud, BEFORE UPLOADING, will mostly make the upload work. 

After the upload works once, the serial monitor can be set to anything and the upload will continue to work, as long as the Nano Every is not powered off. I am  not sure what happens if the reset button is pressed, though.

It would be nice if someone were to integrate this workaround into the the Arduino IDE (I am using version 2.3.2 , on Ubuntu 22.04.04)

Cheers,
B

rph-r, 2024-06-02:
Hello, 
I have this issue for years and years, and I'm working a lot with nanos. On 1.8.19 I simply had to click several times on _Upload_ and finally it was working, until I unplugged the Arduino. 
Now on 2.3.2, IDE get stuck on _uplaoding..._ and I have to quit.
Painful, as you said...

anttiryt, 2024-08-11:
For me resetting the arduino remotely helped. 
I have Nano Everys with sign-on command: status -1 problems. 
Suddenly flash worked in a jif ( sorry my stepper is still not moving ). 

This short script helped me reset Nano Every  ( https://stackoverflow.com/a/55985897/468921 )

    sleep 1
    while true; do    
        echo -en "\x01\x02"    
        sleep 0.1
     done | socat -T1 -t1 - /dev/ttyACM1,hupcl=1,raw,b1000000,cs8,echo=0

w-u-2-o, 2024-08-25:
I'm a brand new Arduino user, but a highly experienced engineer, so not a "newbie". I'm suffering with the same issues here with my very first Arduino board which, unfortunately, is a Nano Every. I can't go to another board, this is replacing a Nano in an existing project and I need the extra memory of the Every.

Environment is Windows 10 Pro, IDE 2.3.2. I tried IDE 2.0.2 but had the same problem. The IDE automatically detects the board and will read board info. 

At this point I'm dead in the water. This seems ridiculous, Arduino development is supposed to be "easy".

Is there no work-around for this issue?

Is there a separate UPDI programmer I can use with the IDE? Perhaps [this](https://www.adafruit.com/product/5879) one?

FWIW, with the verbose option turned on here is the IDE output:

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Radioserver2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): timeout/error communicating with programmer (status -1)

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1
```

anttiryt, 2024-08-25:
> Overriding Baud Rate          : 115200
> avrdude: jtagmkII_getsync(): sign-on command: status -1
> avrdude: jtagmkII_getsync(): sign-on command: status -1
> avrdude: jtagmkII_getsync(): sign-on command: status -1
> avrdude: jtagmkII_getsync(): sign-on command: status -1

FWIW, all my problems with -1 have been that another program is using the COM port. It seems, atleast on Linux, that only one program can access the special USB com port at a time. 

I know you're using windows, but try to look if you can somehow find some other program that's using the COM port. is it possible there's two versions of the software running? Check out also process list and systray. Try uninstalling IDE and going with CLI. Or try uninstalling CLI and going with IDE.

Also, it's not a bad idea to use another computer to program the Arduino. Maybe an old laptop? You just install CLI and use RDP.

w-u-2-o, 2024-08-25:
There are no other programs using the COM port in question. If there was a
conflict then how would the IDE automatically detect the presence of the
board, and how would it be able to read the board info?

w-u-2-o, 2024-08-26:
That work-around does not work for Windows. Apparently that is a Linux work-around only. So I proceeded to try this on the Linux machine I had set up in addition to Windows, as already noted above.

The instructions are not complete. You can't open the serial monitor and set it to 1200 without first connecting the board. So I did that. Then I disconnected and reconnected the board with the serial monitor already open and set to1200.

After that it took 2 tries to upload the sketch. The first try failed about halfway through. The second try appeared to be successful.

gratefulfrog, 2024-08-26:
Sorry for the bad info on the work around. I wasn't in front of my machine.

On Linux that fix works nearly 100% of the time.

I have no idea about development on windows as I seek to be  fully open
source and haven't developed on windows in 30 years...

There must be a solution though because the nano every is a very popular
board.

I hope you find one!

Cheers,
Bob

w-u-2-o, 2024-08-28:
Following up on this: right now it appears that there is some issue with my Windows machine. Linux works perfectly on a separate machine. I'm thinking it is some USB hardware limitation rather than a problem with Windows or drivers.

Stephen-Hazel, 2025-11-17:
I've found this is DEFinitely still an issue on kubuntu 25.10.
 
Can't the Every avrdude conf file set the baudrate to 1200?  I tried setting it in my .avrduderc and it won't take.  The conf is definitely read and parsed (boy does it whine if syntax isn't right:) but the config listing still shows a high baud rate that it uses.

gvnmi, 2026-10-06:
Still a problem in 2026, IDE version 2.3.10, Linux Mint kernel 6.8.0-138.  I have labeled my unboxed Nano Every boards "DO NOT USE" to avoid frustration, it looks like this won't get fixed any time soon :(

A pity, as the Every is a rare 3.3V in a Nano form factor without any extraneous bells and whistles.  I was hoping to use it frequently but I'm designing it out because programming them is such a chore.

The Nano ESP32 is a different story - it cannot have more than 1 board connected at a time or the IDE won't program it.  Sigh.

Experienced engineer.  Using Arduino for rapid prototyping of simple tasks as it's supposed to be easy.