arduino/ArduinoCore-megaavr #125: Arduino Nano Every VCP/USB (SAMD11) fails after reconnecting several times

tehkillerbee, 2022-12-02:
Hello,

I have previously added this issue to https://github.com/arduino/Arduino/issues/11845 but perhaps it is better suited here:

We have noticed a strange issue with the Arduino Nano Every. We are currently using it in a configuration, where it can be powered from either USB or external 5V.

If we unplug/reconnect USB several times in a row (eg. 10 times or less), while the MCU is also powered through external 5V, it will usually stop responding to USB commands.
```
[ 1641.259130] usb 1-1: new full-speed USB device number 87 using xhci_hcd
[ 1641.408817] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1641.408833] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1641.408840] usb 1-1: Product: Arduino Nano Every
[ 1641.408846] usb 1-1: Manufacturer: Arduino LLC
[ 1641.408851] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1641.416271] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1644.161726] usb 1-1: USB disconnect, device number 87
[ 1645.011262] usb 1-1: new full-speed USB device number 88 using xhci_hcd
[ 1645.160488] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1645.160504] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1645.160511] usb 1-1: Product: Arduino Nano Every
[ 1645.160517] usb 1-1: Manufacturer: Arduino LLC
[ 1645.160522] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1645.163561] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1646.644814] usb 1-1: USB disconnect, device number 88
[ 1647.638987] usb 1-1: new full-speed USB device number 89 using xhci_hcd
[ 1647.788649] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1647.788664] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1647.788671] usb 1-1: Product: Arduino Nano Every
[ 1647.788677] usb 1-1: Manufacturer: Arduino LLC
[ 1647.788683] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1647.792096] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1649.255027] usb 1-1: USB disconnect, device number 89
[ 1650.563019] usb 1-1: new full-speed USB device number 90 using xhci_hcd
[ 1650.715927] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1650.715932] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1650.715934] usb 1-1: Product: Arduino Nano Every
[ 1650.715936] usb 1-1: Manufacturer: Arduino LLC
[ 1650.715937] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1650.718352] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1652.019528] usb 1-1: USB disconnect, device number 90
[ 1653.554804] usb 1-1: new full-speed USB device number 91 using xhci_hcd
[ 1653.704118] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1653.704134] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1653.704142] usb 1-1: Product: Arduino Nano Every
[ 1653.704147] usb 1-1: Manufacturer: Arduino LLC
[ 1653.704152] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1653.707254] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1654.947625] usb 1-1: USB disconnect, device number 91
[ 1656.582950] usb 1-1: new full-speed USB device number 92 using xhci_hcd
[ 1656.731956] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1656.731971] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1656.731979] usb 1-1: Product: Arduino Nano Every
[ 1656.731984] usb 1-1: Manufacturer: Arduino LLC
[ 1656.731990] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1656.735186] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1658.118115] usb 1-1: USB disconnect, device number 92
[ 1659.802809] usb 1-1: new full-speed USB device number 93 using xhci_hcd
[ 1659.952061] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1659.952076] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1659.952083] usb 1-1: Product: Arduino Nano Every
[ 1659.952088] usb 1-1: Manufacturer: Arduino LLC
[ 1659.952093] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1659.955226] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1661.226333] usb 1-1: USB disconnect, device number 93
[ 1662.942955] usb 1-1: new full-speed USB device number 94 using xhci_hcd
[ 1663.092296] usb 1-1: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 1663.092303] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1663.092306] usb 1-1: Product: Arduino Nano Every
[ 1663.092308] usb 1-1: Manufacturer: Arduino LLC
[ 1663.092310] usb 1-1: SerialNumber: 79A2E0BE51514A4E434B2020FF0B4633
[ 1664.430570] usb 1-1: can't set config #1, error -71
```
The errors vary, but usually along the following

    can't set config #1, error -71
    device descriptor read/64, error -110

The only solution is to power cycle the 5V supply while having disconnected USB. Pressing the reset button has no effect.

We can see the m4809 is still alive (heartbeat LED) so I assume the SAMD11 is the culprit. Is there a way to reset the SAMD11 from the m4809?

tehkillerbee, 2022-12-19:
Anyone? Perhaps someone with more experience using the SAMD11 could give me some pointers on where to look to investigate this issue further? @facchinm perhaps? 

Anyways I will try to investigate the issue further and let you know if I find a solution :)

facchinm, 2022-12-21:
As per https://github.com/arduino/ArduinoCore-megaavr/issues/124 , it looks like the D11 firmware is not bulletproof, so it might end up crashing on certain situations.
My suggestion to start debugging it would be:
* attach to the D11 using SWD pins (testpoints on the bottom of the board)