arduino/ArduinoCore-megaavr #128: MuxTO UART reception problem

chalupec, 2023-03-17:
Hello, i have loaded successfully last version, i have replaced CRC table with calculation. I have encountered errors on UART reception. 115200baud speed. Transmittion from atsam->atmega works great(perfect timing and levels), also atmega->atsam works (checked by external receiver, perfect timing and levels), but atsam passes invalid data to serial terminal in PC. Basically i am sending "POS 1|1|1|1\r\n" and i get "POScddd>d" in terminal or "Pdcdddkd".. Does anybody encountered something similar?

chalupec, 2023-03-23:
Even if i CRC lookup table is left unchanged and some other parts like JTAG etc, which are not used at that moment, are removed, it will enable me to compile and download firmware, but same problem with corrupted UART receive message occure, so that it is likely that speed of CRC calculation doesnt affect communication.

chalupec, 2023-03-23:
Hello @facchinm did you ever observe something similar, or can you estimate why is this happening?

WestfW, 2025-08-25:
Did you ever have any luck figuring this out?
Changing the CRC algorithm shouldn't make any difference to the serial passthrough mechanism.

chalupec, 2025-08-26:
Hi, didnt make any further investigation or progress. I gave up on this in the past... :-/