arduino/ArduinoCore-megaavr #130: No driver found for ATm32U4DFU / Uno Wifi Rev2

loulouMS, 2023-08-30:
Hello,

The board is not recognized anymore by Windows. It was recognized during several months and from now on, it isn't. But, as the current sketch creates a BLE connection, the device is recognized with Bluetooth. So it seems that's only a problem of USB connection with the computer.

The important point is there is no driver for the device In the Windows device manager. Previously, I think it should be one. When I update the driver, I select the following path: C:\Users\yourUserName\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.8\drivers
or this one: c:\Program Files\Arduino\drivers
In both cases, Windows says it doesn't find any driver for the device.

The board 'Arduino megaAVR boards' is still included in Arduino IDE. There was no change in Arduino configuration except I removed some unused libraries.

Where is and what is the missing driver?

per1234, 2023-08-30:
In order to make all relevant information available to any who are interested in this subject, I'll share a link to the forum topic @loulouMS created about this problem:

https://forum.arduino.cc/t/no-driver-found-for-atm32u4dfu-uno-wifi-rev2/1163568

loulouMS, 2023-09-01:
No solution at the present time from Arduino support.

loulouMS, 2023-09-04:
Finally, the Arduino support proposes a replacement of the board.
As the board is working normally except the upload, I think that the problem is probably due to the USB port of the board.

closed 2023-09-04 by loulouMS