arduino/ArduinoCore-megaavr #134: GCC and Bookworm issues with Arduino UNO WIFI Rev 2 board

steinerrw, 2024-01-15:
After installing all boards/libraries etc with new RPi OS upgrade everything works really smooth. The arduino IDE 1.8.19 runs well...
I can compile and upload for all my different boards except the UNO WIFI Rev2. I have removed board def, reinstalled, rebooted, installed prior versions and nothing seems to get over this error...
It was working before the upgrade.

Adding additional verbose error messages...

Arduino: 1.8.19 (Linux), Board: "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/rick/.arduino15/packages -hardware /home/rick/Arduino/hardware -tools /usr/share/arduino/hardware/tools/avr -tools /home/rick/.arduino15/packages -libraries /home/rick/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=03EB_2145 -ide-version=10819 -build-path /tmp/arduino_build_168707 -warnings=none -build-cache /tmp/arduino_cache_356267 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/rick/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rick/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/rick/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rick/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rick/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rick/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/rick/Arduino/nRF2401-Transmitter/nRF2401-Transmitter.ino
arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/rick/.arduino15/packages -hardware /home/rick/Arduino/hardware -tools /usr/share/arduino/hardware/tools/avr -tools /home/rick/.arduino15/packages -libraries /home/rick/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=03EB_2145 -ide-version=10819 -build-path /tmp/arduino_build_168707 -warnings=none -build-cache /tmp/arduino_cache_356267 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/rick/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rick/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/rick/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rick/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rick/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rick/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/rick/Arduino/nRF2401-Transmitter/nRF2401-Transmitter.ino
Using board 'uno2018' from platform in folder: /home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8
Using core 'arduino' from platform in folder: /home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8
Warning: platform.txt from core 'Arduino megaAVR Boards' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. **Consider upgrading this core.**
Detecting libraries used...
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR   -DMILLIS_USE_TIMERB3 "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/cores/arduino/api/deprecated" "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/cores/arduino" "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/variants/uno2018" "/tmp/arduino_build_168707/sketch/nRF2401-Transmitter.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR   -DMILLIS_USE_TIMERB3 "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/cores/arduino/api/deprecated" "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/cores/arduino" "-I/home/rick/.arduino15/packages/arduino/hardware/megaavr/1.8.8/variants/uno2018" "/tmp/arduino_build_168707/sketch/nRF2401-Transmitter.ino.cpp" -o "/tmp/arduino_build_168707/preproc/ctags_target_for_gcc_minus_e.cpp"
avr-g++: error: device-specs/specs-atmega4809: No such file or directory
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

I've corrected the warning in platform.txt...

I am able to find this file at ...

I was able to search and found the file in question...

/home/rick/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/lib/gcc/avr/7.3.0/device-specs

but am unsure what or where the path is or should be...

facchinm, 2024-01-16:
Hi Rick, for some reason the IDE is picking up ` -tools /usr/share/arduino/hardware/tools/avr` and then using the compiler in /bin . Did you install the IDE via package manager maybe?

steinerrw, 2024-01-16:
Martino;
I installed the IDE via Arduino's software download page and unzipped...
Is there a fix where I can change where that file is picked up from.
Thanks
Rick